fra589/grbl-Mega-5X #79: Spindle enable (laser enable) did not restore after feed hold, posible bug

sinfocomp, 2019-11-17:
Hello, i have upgraded (finally) my laser cutter fron nano to Mega, however if i do a feed hold during a cut (laser enabled via spindle enable and power controlled by spindle pwm) when i do a continue command it starts moving agan but laser off untill another g0 and g1 move again
Thanks for any advice on this

fra589, 2019-11-19:
Hi @sinfocomp,  
  
The normal operation of the hold mode in milling machine is to leave the spindle enabled. So, there is no code to restarting the spindle in the resume command which is the same than the Cycle Start.
In the (relatively new) laser mode, a new option has been added in config.h: 
`#define DISABLE_LASER_DURING_HOLD`
This option will automatically disable the laser during a feed hold by invoking a spindle stop override immediately after coming to a stop. This is a safety feature to ensure the laser doesn't inadvertently remain powered while at a stop and cause a fire.  
  
This explains the behavior that you describe.  
  
I will correct it with an upgrade of the "Cycle Start / Resume" function ASAP.  
  
@++;  
Gauthier.

sinfocomp, 2019-11-19:
Great !, Thanks

sinfocomp, 2020-10-22:
Any update con this ?

fra589, 2020-10-22:
Hi @sinfocomp,  
  
Since I am newly retired since a little bit less than one year, I have much less time available than before :smiley:.  
  
Sunny days were much more conducive to travel, but now I will be able to resume the evolution and corrections of grbl-Mega-5X.
  
You can hope to see this subject evolve before the end of this year...
  
@++;
Gauthier.

truglodite, 2021-02-18:
I am not sure if this has been fixed properly yet, but just commenting out `#define DISABLE_LASER_DURING_HOLD` worked for me. This is when I am operating with laser mode ($32=1) and using M4 for laser control. Not sure how if this would be safe with an M3 file. I am also running an older fork of this repo (enducross), but AFAIK nothing different that would affect this bug.

atomino-max, 2021-12-12:
Hi @sinfocomp,
can you tell me where you have connected the pwm laser input ?
I have defined in defaults.h     #define DEFAULT_LASER_MODE 1 // true, I set $32=1,
but no any pwm output I have checked on D4,D5,D6,D8,D9 and D11 !!!!

Ams-Laser, 2021-12-12:
@atomino-max (im @sinfocomp but lost preious account) im using D6 as laser pwm output on a C02 laser (generic psu from china)

atomino-max, 2021-12-13:
@Ams-Laser  thanks, now it's working on D6, I thought it was possible to have both phisical connections for spindle and laser and switch only with command $32 !!!
Interested in the CO2 laser, can you give me some advice on what power to use and what can be done with it? 
Do you need to refill the CO2?
Many thanks.

fra589, 2021-12-13:
Hi @atomino-max,  
  
Curently, it's not possible to have both spindle and laser connected and switch with grbl command ($32 or other...) it' in the todo list, but I don't know when I will can find the time for this.  
  
@++;
Gauthier.