fra589/grbl-Mega-5X #82: QUESTION where can i find the PWM frequency options for laser control ?

sinfocomp, 2019-11-22:
where can i find the PWM frequency options for laser control ?
looks like if i go higer than 85% (S850) the laser power goes down 
example S900 gives less power than S800, but S500 gives less than S800

fra589, 2019-11-24:
Hello @sinfocomp,  
  
Perhaps you will find reply about PWM ferquency managment in Arduino here:  
https://forum.arduino.cc/index.php?topic=72092.msg541587#msg541587
Or here for full information:  
[Atmel ATmega640/V-1280/V-1281/V-2560/V-2561 Datasheet - Microchip Technology](https://ww1.microchip.com/downloads/en/devicedoc/atmel-2549-8-bit-avr-microcontroller-atmega640-1280-1281-2560-2561_datasheet.pdf)  
  
Then, go to the spindle_control.c file, in spindle_init() function to make your adjustments.  
  
@++;  
Gauthier.

sinfocomp, 2019-11-25:
fixed, closing

closed 2019-11-25 by sinfocomp