fra589/grbl-Mega-5X #84: Enable 6th axis

JeffHank1, 2019-11-24:
I have built a platform with 3 rotational axes - ABC - and want to enable the 6th axis in Grbl Mega 5X. I edited the Config.h file to allow 6 axes and didn't find any other places to change. However after recompiling and uploading, the various GRBL senders I have tried don't recognize the 6th axis and the GRBL Configuration settings have not changed past $134.

Is there more to modify in GRBL Mega 5X beyond the Config.h file to allow the 6th axis to be recognized?
Thanks

fra589, 2019-11-25:
Hi @JeffHank1,  
  
For enabling 6 axis in grbl-Mega-5X, it's enough to put:  
```
  #define N_AXIS 6           // Number of axes
````
in the config.h file.  
  
Don't forget to reset grbl to "factory default" by issuing a "$RST=*" command when you change the axis number. If you don't do this, grbl may have unexpected behaviors.  
  
When I began to write 5, and 6 axis version of grbl, there was no sender available to use more than 4 axis. So, I began to write my own:  
https://github.com/fra589/cn5X.  
It is still in beta, but I already use it for main grbl functionalities. Perhaps you can try it :-)  
  
@++;  
Gauthier

JeffHank1, 2019-11-25:
Thank you for your prompt reply. I tried the Reset command and it did not help. When I ran your CN5X console program and it only showed 5 axes I knew I had done something incorrectly. It turns out I was tripped up by Arduino IDE's Library again. After editing the Config.h file in there and re-uploading, the 6th axis works properly. Grbl Mega 5X is a great version. Thanks again for porting it to the Arduino Mega 2560. Grblgru works very well with it.
I am sure many are waiting for you to finish porting GRBL to the Arduino Due too, lol.

closed 2019-11-25 by JeffHank1