fra589/grbl-Mega-5X #85: Hi

GreySquirrel2002, 2019-11-26:


fra589, 2019-11-26:
?

GreySquirrel2002, 2019-11-27:
Hi 
Sorry, new at this, will try asking for help again (assuming this is the correct forum).
I am trying to make a laser engraver based on an Ardunio mega and Ramps 1.4. After installing grbl-Mega-5X with UGS I am able to move X&Y steppers but any attempt to use Z or others stepper ports,  hangs the system and does not respond to any command. (All hardware appeared to work OK with Marlin.)
Below should be a list of settings  from $$
Any ideas about how I can resolve this problem greatly appreciated.
>>> $$
$0=0
$1=128
$2=0
$3=0
$4=0
$5=0
$6=0
$10=63
$11=0.020
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=10.000
$25=100.000
$26=250
$27=5.000
$30=1000
$31=0
$32=0
$100=250.000
$101=250.000
$102=250.000
$103=500.000
$104=500.000
$110=300.000
$111=300.000
$112=200.000
$113=250.000
$114=250.000
$120=200.000
$121=300.000
$122=0.000
$123=0.000
$124=0.000
$130=-2147483.648
$131=200.000
$132=200.000
$133=200.000
$134=200.000

fra589, 2019-11-28:
Hi @GreySquirrel2002,  

Yes, It may be the good place for help :smiley:   

From your $$ output:  
```
$122=0.000
$123=0.000
$124=0.000
$130=-2147483.648
```
It seems that the eeprom is corrupted by a precedent firmare on your card...
For example, $122 = 0 means the Z Acceleration, in mm/sec^2 is equal to 0 ! So no move it possible...  

After changing firmware from another than grbl-Mega-5X or after changing the number of axis, you need to reset, clears and restores all of the EEPROM data used by Grbl by issuing the $RST=* grbl command.
  
Then, see the [wiki](https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration) to verify and adjust your parameters.  
  
@++;  
Gauthier.

GreySquirrel2002, 2019-11-28:
Hi Gauthier
Thanks for your help.
I now have the system working but had to change config.h 
Line 53  num axis from 5 to 3 before things began to work.
 #define N_AXIS 3            // Number of axes  
But that's OK for my application.
However only the Z axis limit switch is working. Am I right in thinking that the others have not yet been implemented or do I have other problems.
Best Regards Tom

fra589, 2019-11-28:
Hi Tom,  
grbl-Mega-5X is fully functional for 3 to 6 axis with limits switchs on all axis.  
You may still have other problems hardware or logical...  
Compile time parameters are in config.h, electronicals hardware mapping are in cpu_map.h and defaults axis parameters (speed, accel, ...) are in default.h.  
@++;  
Gauthier.

GreySquirrel2002, 2019-11-28:
Hi Gauthier
Thanks for the reply.
I'm not sure hoe or were to begin tackling his problem.
The most obvious place to start was with the hardware. to that end I down loaded and and installed Mega/Ramps code which successfully demonstrated that all the stepper motor drivers were working.
But one thing I did notice that with grbl-Mega-5X installed sending ? using UGS did not return any information (See below).
I wonder if this might be significant in some way?
Many Thanks 
Tom

Grbl 1.1g ['$'
Grbl 1.1g ['$' for help]
[MSG:Check Limits]
[MSG:'$H'|'$X' to unlock]

>>> $X
[MSG:Caution: Unlocked]
ok
>>> $
[HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
ok

>>> ?
ok

GreySquirrel2002, 2019-11-28:
Woops Just read that there is an issues with sending ? with UGS.

So I used Arduino Ser Mon which returned responses as below after sending position changes and a couple ? (s) with Limit switches triggered (links on X max and Z max then removing them and placing a link on Y max.

The results seem to suggest that the  limit switch port is detecting changes but as I have said earlier these changes to limit switch state don't appear to effect program flow!

<Idle|MPos:4.200,3.400,3.800|FS:0,0|Pn:XZ>
ok
----------
Then changing links to y only
------------
<Idle|MPos:4.200,3.400,3.800|FS:0,0|Pn:Y>
ok

(The relevant settings are:-

$20=0
$21=1
$22=1)

Not sure if you are able to make any sense of this but I think it is beyond my capabilities/knowledge to   delve into the various config files.

Many thanks
Tom

fra589, 2019-11-29:
Hi Tom,  
Actually, the hardware limit for grbl-Mega is not implemented. This is due to the pins used by the RAMP board for limits switch doesn't all allow the processor pin change interrupt (https://github.com/gnea/grbl-Mega/issues/97).  
I have writed the new code for enabling hardware limits on RAMPS grbl-Mega version, but it's not already tested, so I can't push it on Github for the moment.  
So, actuazlly, the only way to protect milling machin for preventing travels overflow is to use the software limits ($20 = 1) with good machin max travel configurations ($130 to $135 parameters).  
I should be able to put the new functions on the program online soon, but like any good retiree, I'm always overwhelmed  
@++;  
Gauthier.

GreySquirrel2002, 2019-11-29:
Hi Gauthier
Thanks for the info.
While I don't know enough about the system and how it works can I make the following suggestion for a possible solution.
During "Home" (on my system anyway) the Z axis is driven to Z max end stop, it then moves off releasing the switch. 
Would it not be to possible to perform sequential homing for X & Y using the same end stop pin (if that is the only one which can be used) by linking micro switches.
The set up limits would then be found by the system and saved as the max travel parameters.
Using this approach grbl could then determine its operating limits (find Home on all axes) without the need of a real-time interrupt system (which as I understand it, is where the problem for real-time interrupt lies).
Best regards
Tom

fra589, 2019-11-29:
Hi Tom,  
  
The homing process doesn't need real-time interrupt and it work well for all axis. It is the hard limits security protection stop wich is not implemented for RAMPS hardware.  
  
The homing process parameters are definde in the config.h file (normaly line 171 to 199) and depend of axis number. See the comments from line 156 to 170 (Other comments in this file are also interestings :smile:).  
  
The default behavior when N_AXIS is defined to 3 (line 191 to 193) is to homing Z, then X, then Y. It works normally well.  
  
On standard milling CNC machine, homing is alwais made to the max limit. This normaly leave the maximum space between the tool and the table (piece). Then, all movements are made in negative machine coordinate. It's then necessary need to define G92 coordinate System offset or G54, ..., G59 piece coordinates offset to use positive coordinates movements (See the good Linux CNC doc here: http://linuxcnc.org/docs/html/gcode/coordinates.html#cha:coordinate-system).  
  
@++;  
Gauthier.

closed 2019-12-02 by fra589