fra589/grbl-Mega-5X #87: Homing limited to Z axis only!

GreySquirrel2002, 2019-12-01:
Hi
Thanks for previous help. Now that all my motors are able to function I would like to be able Home each axis.

I have looked at Homing issues experienced by others but none seem to be the same as mine.

Although config.h appears to allow Homing on individual axes, as well all sequentially (H$), only  the Z axis actually Homes! (see below)

Help greatly appreciated.

Settings
$20=0
$21=0
$22=1
$23=0

Home all ($H) - works on Z only:-

Grbl 1.1l ['$' for help]
[MSG:'$H'|'$X' to unlock]
>>> $H
ALARM:8
ok

Home X ($HX)- error 3:-

Grbl 1.1l ['$' for help]
[MSG:'$H'|'$X' to unlock]
>>> $HX
error:3

fra589, 2019-12-01:
Hi @GreySquirrel2002,  

ALARM:8 => Homing fail. Pull off travel failed to clear limit switch. Try increasing pull-off setting or check wiring.
Perhaps an hardware problem or a wrong HOMING_CYCLE definition...

Error 3 on $HX command => You need to enable HOMING_SINGLE_AXIS_COMMANDS in config.h (line 216) by suppress // comments caracter in the begining fo the line.
```
#define HOMING_SINGLE_AXIS_COMMANDS // Default disabled. Uncomment to enable.
```
  
@++;
Gauthier.

GreySquirrel2002, 2019-12-01:
Many thanks for your help Gauthier, I now have things working as required.

Tom

closed 2019-12-01 by GreySquirrel2002