fra589/grbl-Mega-5X #88: Ijk commands (g2/g3) fail on cloned y (y2)

decarpentier, 2019-12-07:
I am having an issue with ijk commands on my cloned y motor (y2), the motor does not energize. The same Goode works fine using Marlin.

fra589, 2019-12-07:
Hi @decarpentier,  
Can you post your config.h (axis definitions, extract lines 50 to 84 and homind definitions 171 to 199) and the $I and $$ output?  
@++;  
Gauthier.

decarpentier, 2019-12-07:
Why would $i show [VER:1.1h.20180712:] when i have downloaded grbl-Mega-5X-1.1l.20190605?

10:36:12.014 -> 
10:36:12.014 -> Grbl 1.1h ['$' for help]
10:36:12.014 -> [MSG:Check Limits]
10:36:12.014 -> [MSG:'$H'|'$X' to unlock]
10:36:17.166 -> [VER:1.1h.20180712:]
10:36:17.166 -> [AXS:5:XYZXY]
10:36:17.166 -> [OPT:VNM,35,255]
10:36:17.166 -> ok

--------------------------------------------------

10:36:59.865 -> $0=10
10:36:59.865 -> $1=255
10:36:59.865 -> $2=0
10:36:59.865 -> $3=21
10:36:59.865 -> $4=0
10:36:59.865 -> $5=0
10:36:59.865 -> $6=0
10:36:59.865 -> $10=3
10:36:59.865 -> $11=0.010
10:36:59.865 -> $12=0.002
10:36:59.865 -> $13=0
10:36:59.865 -> $20=0
10:36:59.865 -> $21=1
10:36:59.865 -> $22=1
10:36:59.865 -> $23=31
10:36:59.865 -> $24=200.000
10:36:59.865 -> $25=2000.000
10:36:59.865 -> $26=200
10:36:59.865 -> $27=1.000
10:36:59.865 -> $30=12000
10:36:59.865 -> $31=0
10:36:59.865 -> $32=0
10:36:59.865 -> $100=100.000
10:36:59.865 -> $101=100.000
10:36:59.865 -> $102=400.000
10:36:59.865 -> $103=100.000
10:36:59.865 -> $104=100.000
10:36:59.865 -> $110=2100.000
10:36:59.865 -> $111=2100.000
10:36:59.865 -> $112=800.000
10:36:59.865 -> $113=2100.000
10:36:59.865 -> $114=2100.000
10:36:59.865 -> $120=50.000
10:36:59.865 -> $121=50.000
10:36:59.865 -> $122=50.000
10:36:59.865 -> $123=50.000
10:36:59.865 -> $124=50.000
10:36:59.899 -> $130=640.000
10:36:59.899 -> $131=342.000
10:36:59.899 -> $132=125.000
10:36:59.899 -> $133=640.000
10:36:59.899 -> $134=342.000
10:36:59.899 -> ok

// Define CPU pin map and default settings.
// NOTE: OEMs can avoid the need to maintain/update the defaults.h and cpu_map.h files and use only
// one configuration file by placing their specific defaults and pin map at the bottom of this file.
// If doing so, simply comment out these two defines and see instructions below.
//#define DEFAULTS_GENERIC
//#define CPU_MAP_2560_INITIAL

// To use with RAMPS 1.4 Board, comment out the above defines and uncomment the next two defines
#define DEFAULTS_RAMPS_BOARD
#define CPU_MAP_2560_RAMPS_BOARD

// Serial baud rate
// #define BAUD_RATE 230400
#define BAUD_RATE 115200

// Axis array index values. Must start with 0 and be continuous.
#ifdef DEFAULTS_RAMPS_BOARD
  // 4, 5 & 6 axis support only for RAMPS 1.4 (for the moment :-)...)
  #define N_AXIS 5            // Number of axes
  #define N_AXIS_LINEAR 5     // Number of linears axis - CHG from 3 to 5 on DEC 5 2019
#else
  #define N_AXIS 3 // Number of axes = 3 if not DEFAULTS_RAMPS_BOARD
#endif

#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V & W.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'

#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'A' // Letter of axis number 4
#endif
#if N_AXIS > 4
  #define AXIS_5 4
  #define AXIS_5_NAME 'B' // Letter of axis number 5
#endif
#if N_AXIS > 5
  #define AXIS_6 5
  #define AXIS_6_NAME 'C' // Letter of axis number 6
#endif
#if N_AXIS > 6
  #error "N_AXIS must be <= 6. N_AXIS > 6 is not implemented."
#endif

// NOTE: Defaults are set for a traditional 3-axis CNC machine. Z-axis first to clear, followed by X & Y.
#ifdef DEFAULTS_RAMPS_BOARD
  #if N_AXIS == 4 // 4 axis : homing
    #define HOMING_CYCLE_0 (1<<AXIS_3) // Home Z axis first to clear workspace.
    #define HOMING_CYCLE_1 (1<<AXIS_4) // Home 4th axis (A)
    #define HOMING_CYCLE_2 (1<<AXIS_1) // Home X axis
    #define HOMING_CYCLE_3 (1<<AXIS_2) // Home Y axis
  #elif N_AXIS == 5 // 5 axis : homing
    #define HOMING_CYCLE_0 (1<<AXIS_3) // Home Z axis first to clear workspace.
    #define HOMING_CYCLE_1 (1<<AXIS_4) // Home 4th axis (A)
    #define HOMING_CYCLE_2 (1<<AXIS_5) // Home 5th axis (B)
    #define HOMING_CYCLE_3 (1<<AXIS_1) // Home X axis
    #define HOMING_CYCLE_4 (1<<AXIS_2) // Home Y axis
  #elif N_AXIS == 6 // 6 axis : homing
    #define HOMING_CYCLE_0 (1<<AXIS_3) // Home Z axis first to clear workspace.
    #define HOMING_CYCLE_1 (1<<AXIS_4) // Home 4th axis (A)
    #define HOMING_CYCLE_2 (1<<AXIS_5) // Home 5th axis (B)
    #define HOMING_CYCLE_3 (1<<AXIS_6) // Home 6th axis (C)
    #define HOMING_CYCLE_4 (1<<AXIS_1) // Home X axis
    #define HOMING_CYCLE_5 (1<<AXIS_2) // Home Y axis
  #else // Classic 3 axis
    #define HOMING_CYCLE_0 (1<<AXIS_3) // Home Z axis first to clear workspace.
    #define HOMING_CYCLE_1 (1<<AXIS_1) // Home X axis
    #define HOMING_CYCLE_2 (1<<AXIS_2) // Home Y axis
  #endif
#else
  #define HOMING_CYCLE_0 (1<<AXIS_3)                // REQUIRED: First move Z to clear workspace.
  #define HOMING_CYCLE_1 ((1<<AXIS_1)|(1<<AXIS_2))  // OPTIONAL: Then move X,Y at the same time.
  // #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable
#endif // DEFAULTS_RAMPS_BOARD

decarpentier, 2019-12-07:
The commits shows that this was fixed on 12-4-2018 in version 1.1i, this would explain why i have this issue still, but i don't understand why the version i downloaded isn't the later version ver 1.1i.

fra589, 2019-12-08:
Hi @decarpentier,
Yes, this should explain... But I don't know which type of mistake can do a download of the 1.1l version from Github and then, upload the 1.1h to the Arduino card...
Perhaps you have more than one version no your computer?
What is your process to upload to Arduino? Do you use Arduino IDE with grblUpload.ino or a separate compil followed by an hex upload with avrdude?
@++;
Gauthier.

decarpentier, 2019-12-08:
Because I am new to this process, I did not delete the old grbl library. Very “rookie” mistake. After doing that I was successful in compiling the latest version. I am unable to test it now as I am leaving for our winter break. Thanks for the quick responses

closed 2019-12-08 by decarpentier