fra589/grbl-Mega-5X #90: Change home cycle sequence

kikirikiki, 2019-12-23:
Hi, I have a 4 axes CNC with GRBL Mega 5X and I would like to change the home cycle sequence. I would like to do the 4th axe homming at the end. How do I have to change the code?
Thanks a lot!

fra589, 2019-12-23:
Hi @kikirikiki,
May be you need to learn reading... The homing cycle is defined in config.h (line 156 and above) with documentation inside the source file !
Search a little bit and you will find.

closed 2019-12-23 by fra589

kikirikiki, 2019-12-23:
Thanks for your answer. I read it but in my machine it starts with the A axis and not with Z axis as what I read in the documentation. That´s why I asked.

[image]

Do I have to change only the line 174 and put it after line 176?