fra589/grbl-Mega-5X #95: 4th axis homing doesn't work !

Flying-Hammer, 2020-01-18:
Hello
I am upgrading a 3 axis CNC,  from Protoneer CNC Shield to Mega+Ramps in order to use a rotating A axis
Quite everything works. Unfortunately rotating axis spins OK but doesn't home
I have followed the pin assignment described on the picture enclosed into the bundle, no way !
Please can somebody tell me where to attach A end-stop or what to change in the config files ?
Regards, and BRAVO to Gauthier Brière for his excellent job

JCG, Lyon, France

fra589, 2020-01-19:
Bonjour @Flying-Hammer,
Entre Français, causons Français 😎 
Pour la définition des branchements, c'est dans cpu_map.h que ça se passe, la config standard est décrite ici : https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout
Peux-tu poster ta configuration d'axes (config.h lignes 50 à 83) et la conf. de prise d'origine "homing" (config.h lignes 171 à 199) ?
@++;
Gauthier.

Flying-Hammer, 2020-01-19:
Bonjour Gauthier
Je venais pour dire "bougez pas, c'est réglé", mais tu avais déjà répondu. Quelle réactivité !
Comme on le constate fréquemment, mon "problème" se situait pile entre la chaise et le clavier : j'avais connecté mes fils au mauvais groupe de pines !
En fait tout marche, les POM, la broche, les boutons, tout.
J'ai fait un proto avec du matos de récupe pour valider le projet, dès demain je fais l'upgrade de la "vraie" machine.
Maintenant qu'on est présentés, si jamais j'ai une VRAIE question, je n'hésiterai pas à te solliciter.
Merci et bien cordialement
JCG

Tu peux si t'as deux minutes regarder là :

[https://www.lesimprimantes3d.fr/forum/topic/27724-projet-grbl-4axes-laser/?tab=comments#comment-324678](url)

fra589, 2020-01-20:
Bonjour Jean-Claude,  
J'ai pris un peu plus de deux minutes pour regarder :smile:  
En réponse aux quelques questions qui se sont posées sur cette page :  
- Sur tous les axes, c'est plutôt le "Max endstop" qu'il faut utiliser... Sur les "vraies" machine, on fait plutôt la prise de références au max car ça dégage au maximum l'espace et normalement, c'est là qu'on est le plus loint de ce qui pourait être abimé... (la pièce déjà montée par exemple). Ce qui fait que si on fait le zéro à cet endroit, toutes les coordonnées machine seront dans l'espace négatif.
- Dans la version originale de Grbl sur carte Arduino Uno, il n'y a qu'une seule connexion pour switch de fin de course par axe qui peut servir a connecter soit le max, soit le min, soit les 2 en série ou en parallèle. Dans la version Arduino Mega + RAMPS, on a choisi d'utiliser les 2 possibilité de connexion min et max, mais elles activent le même bout de code que dans la version Uno. ce qui fait que tu obtiens le homing quelque soit le switch actionné, min ou max. Ça n'a aucune incidence, les deux marchent pareil !
- Si le capteur ne sert "qu'une fois au moment du homing", ce n'est pas parce qu'il est NO, mais c'est parce que les limites matérielles (HARD_LIMITS) ne sont pas activées par défaut. Sur tous les axes, par défaut, les switchs ne sont utilisés que pour le "homing", le contrôle des limites machines étant assuré de manière logiciel pour peu que les dimensions (courses) de la machine aient été correctement entrées.  
Au plaisir,  
@++;  
Gauthier.

closed 2020-01-20 by fra589

LEFOUR, 2021-03-23:
bonjour j'ai essayé de charger le programme mega 5x dans mon arduino mega 2560 mais j'ai une erreur concernant le nombre d'axes erreur C99 la compilation s'arrête et donc pas de chargement dans l'arduino ensuite petit problème avec Xloader se met et update et n'en finit pas merci pour une éventuelle réponse.

fra589, 2021-03-24:
Bonjour,
Merci de ne pas squatter un sujet déjà fermé et qui ne correspond pas du tout à votre problème.
La bonne manière est, après avoir cherché dans les issues précédentes, d'ouvrir sa propre issue.
Ensuite, il faut indiquer suffisamment de détails pour comprendre, ce qui n'est pas le cas ici.
@++;
Gauthier.