fra589/grbl-Mega-5X #96: Choosing whether an axis becomes a linear or rotary axis ?

GrblGru, 2020-01-22:
Hi Gauthier,

first of all, thank you very much for your great work !
I have developed a little CAM, Simulation & Control program for Windows operating systems called 'GrblGru', which I can use to control your Mega-5X.  (See https://youtu.be/zuDK-uluB6g)
I have now been asked by GrblGru users whether I can also control machines with 4 translatory axes or machines with 5 rotary axes. Unfortunately I have no experience in developing programs for the Arduino and so far I only use your current release HEX file.
Is it possible to choose in the configuration whether an axis becomes a linear or a rotary axis, so that you get e.g. 4 translatory axes ?

If this is not the right place for questions of this kind, please point me to the right place.

Thanks for any advice
GrblGru

fra589, 2020-01-22:
Hi @GrblGru,  
  
Yes, this is a good place for questions 😄   
  
The Hex release available on Github is not the last one... I have not yet had time to publish the Hex file for the latest changes I have made...  
But compil is not so dificult on Windows with the [Arduino IDE](https://www.arduino.cc/en/Main/Software) and this procedure : https://github.com/gnea/grbl/wiki/Compiling-Grbl  
  
The axis numbers and the rotary / linear distinction is defined in the config.h file at lines 53 & 54:  
```
  #define N_AXIS 5            // Number of axes
  #define N_AXIS_LINEAR 3     // Number of linears axis
```
And some lines after, their names are defined. See comments in the file.  

All linears axis must be defined before all rotationals axis (the "number" of AXIS_<number> definitions of linears axis must be lower than the number of rotationals axis definitions).  
In the upper example (the standard Hex file definition) there is 5 axis with the 3 first linears and the 2 rotationals at the end.  
  
The $I command of Grbl-Mega-5X have added the [AXS:...] sentence wich tell the axis definition to the interface program. If you respect the standard axis names convention X, Y, Z, U, V, W for linears and A, B and C for rotationals, the interface program will know all he need.  
  
@++;  
Gauthier.

sinfocomp, 2020-01-22:
Hello @grblgru is your software open source?

GrblGru, 2020-01-23:
Thank you very much for your quick and detailed answer. Especially because it is the answer I was hoping for. :)
I will follow your recommendations and try to implement the changes.

@sinfocomp
No, GrblGru is not open source, but it's free.
-> www.grblgru.com

fra589, 2020-01-26:
@GrblGru wrote:
> No, GrblGru is not open source, but it's free.

Too bad...
> Thank you very much for your quick and detailed answer.

You're welcome! 😃 

knowledge is only worth if it is shared ... This is why open source must be preferred.  

@++;
Gauthier.

closed 2020-01-26 by fra589