fra589/grbl-Mega-5X #97: Small error in position A and B axes

MARIOBASZ, 2020-02-06:
Congratulations for the work, Gauthier.
There is a small difference in the position on the A and B axes:
Shipping G0 A100 B100, and the position is A 100.013 B100.013
By the way, sometimes I have to repeat the "?" to visualize Mpos or Wpos.
Hugs

fra589, 2020-02-06:
Hi @MARIOBASZ,
This behavior can be normal...
Depending of the stepper motor definitions and configuration, grbl can only move to an entire number of motor steps. If the position you want is not on a entire step, it will move to the nearest step and report it's exact position.  
If the precision is not enought for you, you need to modify the hardware or refine the microstepping on the stepper driver.  
@++;  
Gauthier.

MARIOBASZ, 2020-02-06:
right, I didn't notice how it was set step / mm

closed 2020-02-06 by MARIOBASZ