fra589/grbl-Mega-5X #100: SPI enhancement

sparkconcepts, 2020-02-18:
Hi fra589, 

Sorry if this is not the right place,  but I was wonder if adding support for SPI drivers was possible.   What would it take? 

Thanks!

fra589, 2020-02-19:
Hi @sparkconcepts,  
  
I have no experience with SPI drivers, but why not will it be possible?  
  
It depend of what do you want to do... ?  
If grbl will be in master or in slave... ?  
Is the new functionality will be soft enough or will it need hard computing... ?  

Arduino Mega 2560 have relatively small hardware capabilities, and grbl ask him a lot of things to control and move the 5 or 6 axis simultaneously while it continue to receiving and calculating next GCodes commands...

@++;  
Gauthier.

miho, 2020-05-25:
Hi, 

it's about configuring SPI capable drivers, such as TMC5160. They allow more complex settings than just microsteps. Without changing the GRBL code it is necessary to use a separate Arduino or other MC to do the SPI configuration of the drivers. It would be really great if we could store those settings in the EEPROM via setting commands (just like velocity and acceleration). When booting up GRBL could send those settings to the drivers. So it is not expensive in terms of CPU. But it would require some PINS for SPI (CLK, MISO, MOSI, up to 5 CS).

closed 2022-06-26 by fra589