fra589/grbl-Mega-5X #103: Duplicate the serial port 0 (USB port) to serial port 2 (pins D16 & D17)

bestiada, 2020-03-10:
Hi Gauthier,
in order to be able to connect and utilize an arduino-based pendant controller, should be a nice idea to "duplicate" the USB port into the serial port 2 (available on pins D16 and D17).
Basically, the system has to work as per follow:

the Mega should be able to receive g-code commands from USB (like it's actually doing!) AND from the serial port 2.
the Mega should be able to send his feedback messages to USB (like it's actually doing!) AND to the serial port 2.
In this way, I can use the USB port to send g-code stream from my PC, exactly like now. But I can also use an arduino-based device to send jog commands and receive feedback. This modification will allow any user to program his own pendant controller, using local push buttons or displays to control and monitor the status of the machine.

If you are interested, I also designed and realized a PCB to connect easily all the necessary hardware, from 3 to 6 axis. The inputs and outputs are opto-isolated and the card is using the same configuration of the Ramps board. The card is still under development but I already have my prototype ready for the tests. If you like to have more info about my card, send me an email at the address bestiada@gmail.com

walterwissmann, 2020-03-10:
Great idea to 'duplicate' USB to Serial to add a pendant controller  +10 ;)

But i think, the backlash routines should come before ;)
https://github.com/fra589/grbl-Mega-5X/issues/40

BR
Walter

fra589, 2020-03-11:
Hi @bestiada
It probably a good idea, but rather bit complicated...
Serials coms are handled by interrupts which have priority over the running code...
So, before we can coding this functionality, we need to define wich priority can have one serial port compared to other...
For example, if a running communication for a long gcode program run on an interface, what do you have to do if another gcode order come on the other interface ?
@++;
Gauthier.  
P.S. @walterwissmann, I don't forget the backlash enhancement :smile:

bestiada, 2020-03-11:
Hi @fra589 
thanks for your prompt replay.
Ok, we can divide the problem in two sections:
1. **Data from external source to Arduino Mega**. I think the priority should go to the serial 0 (USB), since the GCode streaming is mostly coming from there (through PC connection and software like UGS, bCNC or your great CN5X). The serial 2 will be mostly used to connect an arduino-based pendant controller, to ease the preparation of the machine, before to launch the main GCode trough PC and serial 0. Of course, a skilled user can build his own arduino-based pendant controller with the possibility to send the main GCode streaming from there, maybe using an SD card (same like in the 3D printers!). of course, in this particular case, the machine will work on serial 2 only.
2. **Data from Arduino Mega to external source**. Is it possible to just parallel the data? In this case, there is no incoming command to process, just a message to send outside (apologize for my poor knowledge of the software!).

Let me know if my approach sounds good. In any case, I wish to know the opinions/ideas of other users on how to handle the subject...

Regards,
Gianluca

sinfocomp, 2020-09-09:
Hello, this seems a very good idea, here mi two cents, can we totaly disable input 2 while there is comunication going on on usb and only enable after usb is at least 2 or 3 seconds idle

ariefadha, 2023-01-15:
I have idea for this, the goal is to make sure the streaming is safe without any foolproof from user to stream for pendant. and might easier to code.

so the idea :
we need 1 input pin, if that input pin is tied to gnd, the controller use 2nd uart. and pendant can freely stream the gcode (from microsd or button for jogging). 

and if we want to stream for pc (usb) we just tell the pendant to send high signal to the pin.

so if we use internal pull up on the pin, although it loosing signal from pendant, the usb will be the default communication. and the pendant can be activated from menu on display (oled or character display)