fra589/grbl-Mega-5X #104: From 4 to 3 axes

Flying-Hammer, 2020-03-12:
Bonjour à tous, Gauthier
Je viens de finir un graveur laser 4 axes, mais mon axe rotatif n'est pas encore fini de pondre.
Comme le grbl-Mega-5X que j'utilise demande à faire le homing avant de rendre la main, je dois laisser le moteur branché sur la table et actionner le endstop pour "faire semblant" que c'est OK
C'est pas bien gênant mais je voudrais passer temporairement le nombre d'axes de 4 à 3

>   // 4, 5 & 6 axis support only for RAMPS 1.4 (for the moment :-)...)
>   #define N_AXIS 3            // Number of axes
>  #define N_AXIS_LINEAR 3     // Number of linear axis

Quelque chose m'échappe : si je fais ça l'axe X et l'axe Z bougent ensemble !

Une idée ?

Merci d'avance

Jean-Claude Garnier

$i
[VER:1.1m.20191215:]
[AXS:4:XYZA]
[OPT:VNMH,35,255,50]

$$
$0=10 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.020 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=1 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=0 (Homing direction invert, mask)
$24=100.000 (Homing locate feed rate, mm/min)
$25=1000.000 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=2.000 (Homing switch pull-off distance, millimeters)
$30=1000 (Maximum spindle speed, RPM)
$31=1 (Minimum spindle speed, RPM)
$32=1 (Laser-mode enable, boolean)
$100=80.150 (X-axis travel resolution, step/mm)
$101=80.150 (Y-axis travel resolution, step/mm)
$102=1581.000 (Z-axis travel resolution, step/mm)
$103=750.000
$110=1000.000 (X-axis maximum rate, mm/min)
$111=1000.000 (Y-axis maximum rate, mm/min)
$112=500.000 (Z-axis maximum rate, mm/min)
$113=1000.000
$120=100.000 (X-axis acceleration, mm/sec^2)
$121=100.000 (Y-axis acceleration, mm/sec^2)
$122=100.000 (Z-axis acceleration, mm/sec^2)
$123=100.000
$130=352.000 (X-axis maximum travel, millimeters)
$131=297.000 (Y-axis maximum travel, millimeters)
$132=106.000 (Z-axis maximum travel, millimeters)
$133=360.000

fra589, 2020-03-12:
Bonsoir Jean-Claude,  
Ah, c'est bien la qu'il faut poser ta question 😄 
Pour vérifier, je vais (rapidement ?) reproduire la config que tu me dis (`#define N_AXIS 3 & #define N_AXIS_LINEAR 3`) pour vérifier...
Mais au premier coup d'oeuil, il semble que les definition d'axes que tu dis (`#define N_AXIS 3 & #define N_AXIS_LINEAR 3`) n'ont pas été téléversées correctement dans ton Arduino :  
Si on lit le résultat de $i, on trouve :  
`[AXS:4:XYZA]`
Ce qui indique un `#define N_AXIS 4`...
Il faut que tu vérifie que tu as bien recompilé et envoyé grbl avec les bonnes options.  
Ensuite, en cas de changement du nombre d'axes, il est important de réinitialiser les paramètres de l'EEPROM en "config usine" avec la commande : `$RST=*`
@++;
Gauthier.

Flying-Hammer, 2020-03-12:
Merci pour ta rapide réponse
Si le résultat de la commande $i renvoie **[AXS:4:XYZA]** c'est que je suis revenu à la config qui marche en attendant mieux ! (avec le moteur au bout de son fil)
C'est en mettant **#define N_AXIS 3** que ça avait tout planté.

Je ne connaissais pas la commande $RST=*, dès demain je re-tente et je te tiens au coran.

J'aurai aussi des infos à offrir concernant la gestion du laser par PWM : j'ai bénéficié d'une grande assistance de la part de https://github.com/Ingwie  qui m'a autorisé à les partager
Bonne soirée
L'JCG

Le machin ici[ >>>>](https://www.lesimprimantes3d.fr/forum/topic/27724-projet-grbl-4axes-laser/page/4/?tab=comments#comment-335800)

fra589, 2020-03-13:
Bonjour Jean-Claude,
> Je ne connaissais pas la commande $RST=*
La doc de grbl sur le Wiki du projet original, c'est ici: 
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands
Toujours intéressant à lire...
@++;
Gauthier.

Flying-Hammer, 2020-03-13:
Tu avais raison, la commande $RST=* m'a réglé le problème.
Quand mon 4ème axe sera fini, je saurai m'en souvenir
Merci encore

L'JCG

closed 2020-03-13 by fra589