fra589/grbl-Mega-5X #105: What programs can I use this firmware with?

Peter6891, 2020-03-20:
I previously used the **Candle** program. (https://github.com/Denvi/Candle)

The latest version of GRBL firmware, if I'm not mistaken, was 1.1F
I understand that you created the firmware based on GRBL, but for 5 axes ...

Question: if I flash the arduino mega 2560 with your firmware for 5 axes, then which program on the PC can I manage it?

Thanks.
(if I misunderstood something, correct me, please)

fra589, 2020-03-21:
Hi @Peter6891,  
The latest version from the [original Grbl firmware for Arduino Mega](https://github.com/gnea/grbl-Mega) is 1.1g.  
This firmware, grbl-Mega-5X, is in version 1.1m can drive up to 6 axis capability.  
You can use my own open source software : [cn5X++](https://github.com/fra589/cn5X). It is under active developpement but I allready use it for driving a real milling machin.  
It is a basic control panel, but it is written in connection with grbl-Mega-5X with goal of implementing and optimizing all his functionalities.
You can also use [GrblGru](https://www.grblgru.com/), a free program, but not open source. GrblGru made special work to ensure good grbl-Mega-5X compatibility with his software.  
See : https://www.youtube.com/watch?v=zuDK-uluB6g
@++;  
Gauthier.

Flying-Hammer, 2020-03-26:
Hello @Peter6891 
If you are under Windows, you can also use CNCjs which fullfills all my needs
https://github.com/cncjs/cncjs/wiki/Electron-App
@++
JCG

Peter6891, 2020-03-26:
and where you can see the pinout of the ARDUINO MEGA 2560 board for use
with a 5-6 axis CNC

fra589, 2020-03-27:
Quite simply in the grbl-Mega-5X wiki:  
https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout  
@++;  
Gauthier.

closed 2020-04-07 by fra589