fra589/grbl-Mega-5X #106: Homing only 1 motor per axis

miserovitch, 2020-03-23:
Hi,
I m having some troubles to setup grbl-mega-5X on my MPCNC.

All movements are alright but I can't get homing to work. When I home only 1 motor per axis is moving.
In config.h I have 2 axis per linear axis.
`  #define HOMING_CYCLE_0 ((1<<AXIS_2)|(1<<AXIS_5))  // HomeY
    #define HOMING_CYCLE_1 ((1<<AXIS_1)|(1<<AXIS_4))   //HomeX` 

When I am close enough to endstops and declare all axis separately it works
`#define HOMING_CYCLE_0 (1<<AXIS_1)//HomeX
#define HOMING_CYCLE_1 (1<<AXIS_4)//HomeX
#define HOMING_CYCLE_2 (1<<AXIS_2)// HomeY
#define HOMING_CYCLE_3 (1<<AXIS_5)// HomeY`
All endstops are working.

But no way to make it work by 2.

My status report from GRBL also shows that endstop are working normally.

Any of you already had this issue? 

Thanks for your support and great job!

LC

fra589, 2020-03-23:
Hi @miserovitch,  
I don't had already have this issue.  
Can you post your axis definitions from config.h, line 50 to 84, and the full homing configuration alos from config.h, lines 171 to 199 ?  
In addition, you can also post the result of `$I` and `$$` grbl commands...  
@++;
Gauthier.

miserovitch, 2020-03-24:
Hi @fra589 ,
Thanks for your quick reply.
Here are the data requested
Axis definitions 
`// Axis array index values. Must start with 0 and be continuous.
#ifdef DEFAULTS_RAMPS_BOARD
  // 4, 5 & 6 axis support only for RAMPS 1.4 (for the moment :-)...)
  #define N_AXIS 5            // Number of axes
  #define N_AXIS_LINEAR 3     // Number of linears axis
#else
  #define N_AXIS 3 // Number of axes = 3 if not DEFAULTS_RAMPS_BOARD
#endif

#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X'
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'

#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'X' // Letter of axis number 4
#endif
#if N_AXIS > 4
  #define AXIS_5 4
  #define AXIS_5_NAME 'Y' // Letter of axis number 5
#endif
#if N_AXIS > 5
  #define AXIS_6 5
  #define AXIS_6_NAME 'C' // Letter of axis number 6
#endif
#if N_AXIS > 6
  #error "N_AXIS must be <= 6. N_AXIS > 6 is not implemented."
#endif
`
Homing configuration: in my file it is from line 146 to 185.
`#ifdef DEFAULTS_RAMPS_BOARD
  #if N_AXIS == 4 // 4 axis : homing
    #define HOMING_CYCLE_0 (1<<AXIS_4) // Home 4th axis (A)
    #define HOMING_CYCLE_1 (1<<AXIS_1) // Home X axis
    #define HOMING_CYCLE_2 (1<<AXIS_2) // Home Y axis
    #define HOMING_CYCLE_3 (1<<AXIS_3) // OPTIONAL: Home Z axis
  #elif N_AXIS == 5 // 5 axis : homing
    
    #define HOMING_CYCLE_0 ((1<<AXIS_2)|(1<<AXIS_5))  // HomeY
    #define HOMING_CYCLE_1 ((1<<AXIS_1)|(1<<AXIS_4))   //HomeX
    //#define HOMING_CYCLE_2 (1<<AXIS_3) // Home Z axis

  #elif N_AXIS == 6 // 6 axis : homing
    #define HOMING_CYCLE_0 (1<<AXIS_4) // Home 4th axis (A)
    #define HOMING_CYCLE_1 (1<<AXIS_5) // Home 5th axis (B)
    #define HOMING_CYCLE_2 (1<<AXIS_6) // Home 6th axis (C)
    #define HOMING_CYCLE_3 (1<<AXIS_1) // Home X axis
    #define HOMING_CYCLE_4 (1<<AXIS_2) // Home Y axis
    #define HOMING_CYCLE_5 (1<<AXIS_3) // OPTIONAL: Home Z axis
  #else // Classic 3 axis
    #define HOMING_CYCLE_0 (1<<AXIS_1) // Home X axis
    #define HOMING_CYCLE_1 (1<<AXIS_2) // Home Y axis
    #define HOMING_CYCLE_2 (1<<AXIS_3) // OPTIONAL: Home Z axis
  #endif
#else
  #define HOMING_CYCLE_0 (1<<AXIS_3)                // REQUIRED: First move Z to clear workspace.
  #define HOMING_CYCLE_1 ((1<<AXIS_1)|(1<<AXIS_2))  // OPTIONAL: Then move X,Y at the same time.
  // #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable
#endif // DEFAULTS_RAMPS_BOARD

// NOTE: The following are two examples to setup homing for 2-axis machines.
// #define HOMING_CYCLE_0 ((1<<AXIS_1)|(1<<AXIS_2))  // NOT COMPATIBLE WITH COREXY: Homes both X-Y in one cycle.

// #define HOMING_CYCLE_0 (1<<AXIS_1)  // COREXY COMPATIBLE: First home X
// #define HOMING_CYCLE_1 (1<<AXIS_2)  // COREXY COMPATIBLE: Then home Y

// Number of homing cycles performed after when the machine initially jogs to limit switches.
// This help in preventing overshoot and should improve repeatability. This value should be one or
// greater.
#define N_HOMING_LOCATE_CYCLE 1 // Integer (1-128)
`
Reports:
`> $I
[VER:1.1h.20180712:]
[AXS:5:XYZXY]
[OPT:VNM,35,255]`

`> $$
$0=10 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=3 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=0 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$10=2 (Status report options, mask)
$11=0.020 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=1 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=31 (Homing direction invert, mask)
$24=100.000 (Homing locate feed rate, mm/min)
$25=2000.000 (Homing search seek rate, mm/min)
$26=250 (Homing switch debounce delay, milliseconds)
$27=1.000 (Homing switch pull-off distance, millimeters)
$30=1000 (Maximum spindle speed, RPM)
$31=0 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$100=100.000 (X-axis travel resolution, step/mm)
$101=100.630 (Y-axis travel resolution, step/mm)
$102=400.000 (Z-axis travel resolution, step/mm)
$103=100.000
$104=100.630
$110=4500.000 (X-axis maximum rate, mm/min)
$111=4500.000 (Y-axis maximum rate, mm/min)
$112=800.000 (Z-axis maximum rate, mm/min)
$113=4500.000
$114=4500.000
$120=50.000 (X-axis acceleration, mm/sec^2)
$121=50.000 (Y-axis acceleration, mm/sec^2)
$122=50.000 (Z-axis acceleration, mm/sec^2)
$123=50.000
$124=50.000
$130=450.000 (X-axis maximum travel, millimeters)
$131=280.000 (Y-axis maximum travel, millimeters)
$132=80.000 (Z-axis maximum travel, millimeters)
$133=450.000
$134=280.000`

I hope you can help me out.

ps: Un compatriote vivant a Singapour.

LC

miserovitch, 2020-03-24:
`> $$
$0=10 (Step pulse time, microseconds)

$1=255 (Step idle delay, milliseconds)

$2=0 (Step pulse invert, mask)

$3=3 (Step direction invert, mask)

$4=0 (Invert step enable pin, boolean)

$5=0 (Invert limit pins, boolean)

$6=0 (Invert probe pin, boolean)

$10=2 (Status report options, mask)

$11=0.020 (Junction deviation, millimeters)

$12=0.002 (Arc tolerance, millimeters)

$13=0 (Report in inches, boolean)
$20=0 (Soft limits enable, boolean)
$21=1 (Hard limits enable, boolean)

$22=1 (Homing cycle enable, boolean)

$23=31 (Homing direction invert, mask)

$24=100.000 (Homing locate feed rate, mm/min)

$25=2000.000 (Homing search seek rate, mm/min)

$26=250 (Homing switch debounce delay, milliseconds)

$27=1.000 (Homing switch pull-off distance, millimeters)

$30=1000 (Maximum spindle speed, RPM)

$31=0 (Minimum spindle speed, RPM)

$32=0 (Laser-mode enable, boolean)

$100=100.000 (X-axis travel resolution, step/mm)

$101=100.630 (Y-axis travel resolution, step/mm)

$102=400.000 (Z-axis travel resolution, step/mm)

$103=100.000

$104=100.630

$110=4500.000 (X-axis maximum rate, mm/min)

$111=4500.000 (Y-axis maximum rate, mm/min)

$112=800.000 (Z-axis maximum rate, mm/min)

$113=4500.000

$114=4500.000

$120=50.000 (X-axis acceleration, mm/sec^2)

$121=50.000 (Y-axis acceleration, mm/sec^2)

$122=50.000 (Z-axis acceleration, mm/sec^2)

$123=50.000

$124=50.000

$130=450.000 (X-axis maximum travel, millimeters)

$131=280.000 (Y-axis maximum travel, millimeters)

$132=80.000 (Z-axis maximum travel, millimeters)

$133=450.000

$134=280.000
ok`

fra589, 2020-03-24:
Hi @miserovitch,  Bonjour à Singapour 😃 

_Pour la défense de la langue française :_  
Au niveau de la configuration, tout semble correct.?.?.?
A la vue du résultat de $I, il semble que ta version est bien trop ancienne...
Certains bugs (notamment concernant le "homing") ont été corrigés depuis en plus des fonctionalités ajoutées.
La version actuelle à jour est la v1.1m.20191215. je te conseil de mettre à jour ta version et de re-tester.
Au passage, il est important de bien réinitialiser les paramètres avec la commande Grbl : `$RST=*` en cas de changement de version ou de définition des axes.
En l'absence de cette réinitialisation, il arrive fréquemment un comportement erratique.

_Et pour nos amis qui ne comprendraient pas la langue de Molière :_ 
In terms of configuration, everything seems correct.?.?.?
In view of the result of $ I, it seems that your version is far too old ...
Some bugs (notably concerning "homing") have been fixed since in addition to the added functionalites.
The current up to date version is v1.1m.20191215. I advise you to update your version and re-test.
By the way, it is important to properly reset the parameters with the Grbl command: `$ RST = *` in case of change of version or definition of the axes.
In the absence of this reset, erratic behavior frequently occurs.  
  
@++;  
Gauthier.
P.S. J'ai édité ton post pour y placer de "vrai" blocs de codes sur plusieurs lignes. Il faut utiliser des triples marques de code (comme ça : \`\`\` ) sur la ligne avant et sur la ligne après le bloc de code.

miserovitch, 2020-03-25:
Hi @fra589 ,

So i downloaded and reinstalled grbl from this github, I did the $RST=* before key in parameters.
Good news! All axis are moving for homing :-)

Bad news, endstops are not working anymore

Report when $5=0 and endstops not triggered
```
?
<Alarm|WPos:5.000,4.999,0.000,5.000,5.000|Bf:35,254|FS:0,0|Pn:XYXY|Ov:100,100,100>
ok
```
When triggered

```
?
<Sleep|WPos:5.000,4.999,0.000,5.000,5.000|Bf:35,255|FS:0,0|Pn:XYXY>
>
```

$5=1 and not triggered
```
?
<Alarm|WPos:5.000,4.999,0.000,5.000,5.000|Bf:35,254|FS:0,0|Pn:Z>
ok
```
triggered
```
?
<Alarm|WPos:5.000,4.999,0.000,5.000,5.000|Bf:35,254|FS:0,0|Pn:Z>
ok
```

Triggered or not triggered it seems state does not change.
Here the $I report
```
$I
[VER:1.1m.20191215:]
[AXS:5:XYZXY]
[OPT:VNMH,35,255,88]
```

My config file:
```
// Axis array index values. Must start with 0 and be continuous.
#ifdef DEFAULTS_RAMPS_BOARD
  // 4, 5 & 6 axis support only for RAMPS 1.4 (for the moment :-)...)
  #define N_AXIS 5            // Number of axes
  #define N_AXIS_LINEAR 3     // Number of linears axis
#else
  #define N_AXIS 3 // Number of axes = 3 if not DEFAULTS_RAMPS_BOARD
#endif

#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V & W.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'

#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'X' // Letter of axis number 4
#endif
#if N_AXIS > 4
  #define AXIS_5 4
  #define AXIS_5_NAME 'Y' // Letter of axis number 5
#endif
```
```
#elif N_AXIS == 5 // 5 axis : homing
    //#define HOMING_CYCLE_0 (1<<AXIS_3) // Home Z axis first to clear workspace.
    #define HOMING_CYCLE_0 ((1<<AXIS_2)|(1<<AXIS_5)|(1<<AXIS_1)|(1<<AXIS_4)) // Home XY axis

```
```
#ifdef DEFAULTS_RAMPS_BOARD
  // Only enable the following line if you have - (min) limit switches attached
  #define INVERT_MIN_LIMIT_PIN_MASK ((1<<AXIS_1)|(1<<AXIS_2)|(1<<AXIS_4)|(1<<AXIS_5))
  // Only enable the following line if you have + (max) limit switches attached
  //#define INVERT_MAX_LIMIT_PIN_MASK ((1<<AXIS_1) | (1<<AXIS_2) | (1<<AXIS_3))
#endif
```
Sorry to disturb again.
LC

fra589, 2020-03-25:
Hi @miserovitch,  
Try with all the `#define INVERT_<MIN/MAX>_LIMIT_PIN_MASK` commented:  
```
#ifdef DEFAULTS_RAMPS_BOARD
  // Only enable the following line if you have - (min) limit switches attached
  //#define INVERT_MIN_LIMIT_PIN_MASK ((1<<AXIS_1)|(1<<AXIS_2)|(1<<AXIS_4)|(1<<AXIS_5))
  // Only enable the following line if you have + (max) limit switches attached
  //#define INVERT_MAX_LIMIT_PIN_MASK ((1<<AXIS_1) | (1<<AXIS_2) | (1<<AXIS_3))
#endif
```
This is only for specials case when all your limit are not same, normaly open or normaly closed.

If I understand fine, I think your limits switch are on the min direction... Am I right?  
But it stay a question for me, is your switchs are normaly open or normaly closed?  

@++;  
Gauthier.

Flying-Hammer, 2020-03-26:
Hello Miserovitch
Toi ici quelle coïncidence....
Je suis sûr que l'ami Gauthier va te tirer d'affaire, moi il m'a bien aidé avec son $RST* !
JCG

fra589, 2020-03-26:
Hi,
> @Flying-Hammer wrote:
>  Je suis sûr que l'ami Gauthier va te tirer d'affaire, moi il m'a bien aidé avec son $RST* !

Ouaips ! mais là, ça va peut-être prendre un tout petit peu plus de temps...
Je viens de revoir le code concerné (dans limits.c) et c'est bien tordu comme code ! Il faut vraiment que je corrige et simplifie ça.
@++;
Gauthier.
P.S. Attention, c'est `$RST=*` ne pas oublier le "="

miserovitch, 2020-03-27:
Hehe le monde est petit en effet.

C'est super sympa @fra589 ! Prends ton temps, dans l'intervalle ce sera moins complique que de mon cote je passe les endstops en NO.
J'en profite pour une question de newbie, peut-on charger les parametres en masse ou fait-il les saisir ligne a ligne a chaque compilation du firmware? Peut-on les inclure dans le firmware pour que cela devienne les parametres de bases qui sont charges avec un '$RST=*' ?

@+

Laurent

Flying-Hammer, 2020-03-27:
Je n'ai pas la prétention de répondre à la place de Gauthier, mais pour une mystérieuse raison qui m'échappe, ça a l'air de garder les paramètres d'une fois sur l'autre.
Sauf si on change lourdement la config (passage de 4 à 3 axes comme j'ai dû le faire momentanément) là j'ai dû tout réinitialiser.

fra589, 2020-03-27:
Bonjour Laurent, bonjour Jean-Claude,  
  
Pour charger les paramètres en masse, il suffit de les placer tous dans un fichier gcode, un par ligne,  sous la forme `$X=<Valeur>`.  
  
Pour les inclure dans le firmware afin que ça soit les valeurs par défaut réinitialisées par `$RST=*`, il suffit d'éditer le fichier defaults.h et de mettre les valeurs désirées dans le bon groupe de paramètres (dans la section #ifdef DEFAULTS_RAMPS_BOARD). Le nom des variables en fonction du paramètre se trouve ici :  https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration.
  
Pour ta culture, Jean-Claude, la raison qui permet aux paramètres (et a d'autres variables) d'être retenues d'une fois sur l'autre est que toutes ces valeurs sont écrites en mémoire EEPROM et que la mémoire EEPROM n'est pas effacée lors du flashage d'un nouveau code. Le code est stocké dans la mémoire FLASH qui n'est pas la même que la mémoire EEPROM.
Ce qui fait que le fonctionnement devient erratique quand on change le programme, c'est que les paramètres sont stockés en EEPROM bout à bout dans l'ordre et non avec leur nom. Les nom de variables sont totalement inconnus après compilation d'un programme, seules restent des adresses dans les différentes mémoires. Ce qui fait que si la taille ou le nombre des paramètres change en fonction d'une nouvelle compilation, le programme prendra ce qu'il trouve à telle adresse comme telle valeur, sans savoir qu'en fait, c'est l'adresse qui correspond à un autre paramètre de la version précédente du programme...
Ça peut lui faire bizarre ! 😈 
Le code de `$RST=*` permet de forcer la ré-écriture de toute la mémoire EEPROM avec les valeurs par défaut définies dans defaults.h et donc de remettre tous les paramètres à leurs place.  
  
Allez, un petit lien pour votre culture : https://zestedesavoir.com/tutoriels/374/gestion-de-la-memoire-sur-arduino/ 😃   
  
@++;  
Gauthier.

Flying-Hammer, 2020-03-27:
Cool, "La connaissance est la seule chose qui s'accroît quand on la partage"
Ahlala, ce @fra589  s'il existait pas, faudrait l'inventer

Prenez soin de vous les gars....

hhovde, 2020-03-27:
I'm experiencing the same issues as @miserovitch describes. 
(I initially followed @RaphaelDives instructions here: https://github.com/fra589/grbl-Mega-5X/issues/4#issuecomment-450506340)

miserovitch, 2020-03-28:
Hi Guys,

Thanks for the effort but I give up. 
With the new firmware the 4 motors are moving for homing that's right but not at the same feedrate, so after few centimeters it blocks.

For endstops, now they are all normally open, I check with a tester that loop is closed when endstops are triggered. I checked on connectors to plug on ramps so no wiring issue.
Only 2 of them are recognized now (1 on each axis). IF I change the firmware to activate   

```
#define INVERT_MIN_LIMIT_PIN_MASK ((1<<AXIS_1)|(1<<AXIS_2)|(1<<AXIS_4)|(1<<AXIS_5))
```
When not triggered the firmware detect them as all closed (normal in that case), when triggered only 2 them goes to open status in the report. 

I'm desperate, I go back to Marlin for the moment. Probably not as good as GRBL for CNC but at least for me it works. 
I am fully aware that problem is situated between the screen and the chair.
I'll probably redo a try in close future.

Thanks @fra589 for your quick (and detailed) response.

@+

Laurent

Flying-Hammer, 2020-03-28:
Hi Laurent
As I told you, U can have a second micro-SD ready for Marlin waiting for Gauthier's homework.
Fortunately, he has a lot of time to find the solution : RC gliders are confined !
Have a nice weekend (so far "weekend" has a meaning in these circumstances) and take care of yourself

fra589, 2020-03-30:
Hi @miserovitch,  
I found this is wrong in your config.h:  
`#define N_AXIS_LINEAR 3     // Number of linears axis`
As all your axis are linear, you need to do:  
`#define N_AXIS_LINEAR 5     // Number of linears axis`
I'm not shure, but this can explain the speed difference you noticed.
@++;  
Gauthier.

hhovde, 2020-03-30:
@fra589 
I have N_AXIS_LINEAR set to 5, and it does not resolve the problem..

As far as I can tell, the problem seems to be related to the cloned axis and the bitmasks set in cpu_map ?

fra589, 2020-03-30:
Hi @hhovde, 

> I have N_AXIS_LINEAR set to 5, and it does not resolve the problem..  

That's what I feared :disappointed: :persevere:  
I will need to make an hardware test platform to reproduce the behavior...
This will make take a while... :new_moon: :waxing_crescent_moon: :first_quarter_moon: :waxing_gibbous_moon: :full_moon: :waning_gibbous_moon: :last_quarter_moon: :waning_crescent_moon: :new_moon:
But I will find !
@++;  
Gauthier.

miserovitch, 2020-03-31:
Hi,

What is really strange is that at first I followed the tuto [ZOOM grbl+RAMPS+auto squaring](https://alexpgh.github.io/foss-toolchain-mpcnc/zoom_1/) that relates to GRBL mega 5X  [johnboiles/grbl-Mega-5X](https://github.com/johnboiles/grbl-Mega-5X/tree/c640018f846bba11a4228c5758af2fb21367d01a) which a fork of your own github @fra589 .

In this version I had movements OK, endstops NC OK, homing NOT OK at all.

In your version moveemnts are OK, endstops NC NOT OK, homing that move but not a the same speed. 
So endstop NC issue has been solved on the other fork.

On your version @fra589 I tried X and Y together, not one after the other...
I will reflash your GRBL to check this case. Maybe one after the other can solve this. 

BR
Laurent

fra589, 2020-03-31:
Hi Laurent,  
  
Both NC and NO switch worked in the original grbl-Mega version... But although I suspect the configuration of the mask definition of different switch NO & NC to be buggy.
I had to touch up the limits code to be able to implement the hardware limits, and in doing so, I probably implemented new bug(s)... :dizzy_face: 
I have to review this part. :construction_worker:  
  
I also think that, pheraps, your homing wrong speed is due to homing both X & Y (so 4 axis) at the same time... The speed and acceleration calculation algorithm probably does not take this scenario into account as it use trigonometry calculation up to 3 axis (3 dimensions) to calculate the hipothenus distance of the diagonal of the parallelepiped and I did not test the result in the 4th dimension :interrobang: :smile:  
Tell me your result about homing X and Y separately.  
  
As I tell you before, I need some hardware supply to make a test prototype. :hourglass_flowing_sand:  
  
@++;  
Gauthier.

derFrickler, 2020-04-09:
Hi Gaulthier, thanks for the great work on grbl-MEGA-5x. Using grbl on Uno since years and on ESP32 since some time. This is the firs time i am using it on a MEGA and i am having a quite similar issue as this one.

I am using a RAMPS 1.4 compatible board with 2 Motors on the Y axis and trying to setup the homing. I changed the homing cycle to home all axis separately, well, both y motors at the same time. i  enabled ENABLE_RAMPS_HW_LIMITS

#define N_AXIS 4            // Number of axes             
#define N_AXIS_LINEAR 4     // Number of linears axis
#define AXIS_4_NAME 'Y' // Letter of axis number 4  
#define HOMING_CYCLE_2 ((1<<AXIS_2)|(1<<AXIS_4)) // Home Y axis 

Movement works well on all axis, switches report correctly if triggered: 
[verbose] <Alarm|WPos:234.011,-180.482,-5.000,-180.482|FS:0,0|Pn:YY>
I am using switches that close to ground when triggered, so no change needed in grbl.

But one of the Y Motors is not stopping during y homing, even if both of the y the limit switch have been triggered. 

Any hints on how to debug this?

Thanks 
      Johannes

fra589, 2020-04-09:
Hi @derFrickler,  
 
> But one of the Y Motors is not stopping during y homing,

Wich on is not stopping? The one on axis 2 or the one on axis 4?
Are you sure you does not have the limits switchs inverted between axis 2 and 4?
Make sure than you are in the latest version (1.1m.20191215) since this is the one wich I work on.

@++;
Gauthier.

derFrickler, 2020-04-09:
Oh man, checked everything again and found the issue.
One side had a loose belt which was jumping over the pulley when trying to square the portal. Moving worked fine, but while squaring it had no force to reach the limit switch, so the motor keept turning.

Sorry for bothering.

miserovitch, 2020-04-10:
Hello guys,

Sorry for late feedback, reload the grbl firmware with homing one after the other.
Unfortunately no change, the 2 motors do not seem to go at the same speed.
Otherwise no problem for movements, even 500mm are done normally, same for X+Z movements together.

BR

Laurent

fra589, 2020-04-13:
Hi @miserovitch,  
That you say is very strange!  
I have re-checked the homing function code, and all the axes are driven with the same speed, as long as they have the same configuration (same step / unit, same max speed, same acceleration, and same stroke max).
I think you have reset with $RST=* after your last flash of grbl-Mega-5X and re-implement your parameters, can you verify the result of $$ that your cloned axes have all exactly the same parameters?
@++;  
Gauthier.

fra589, 2020-04-24:
Hi @miserovitch,  
I finaly found and corrected the bug in the limits_get_state() function. :smile:  
I jus pushed the corrected version on github (v1.1n.20200424) wich allow using Normaly Closed limit switchs.
@++;
Gauthier.

hhovde, 2020-04-24:
@fra589 That sounds fantastic!
I'll test the new patch over the weekend :)

hhovde, 2020-04-27:
Hi again @fra589
I'm sorry to inform you that the patch does not seem to work. It might be due to my ignorance, but there seems to be something weird going on.
I have tested and tweaked the settings to try to get it to work, but to no avail.
Sometimes limit switches doesn't work (register), other times the machine runs the motors in opposite directions while homing, but works fine while jogging. Other times only one motor is homing.
(I did erase the EEPROM and started with a clean build)

I'm prepared to spend some time and run a series of test to find out what's wrong. But before I do, I would like you to point out what settings I should use to expect pr-axis-homing with NC endstops. This way I at least can eliminate a potential misunderstanding of the basic concepts.

/H

fra589, 2020-04-27:
Good evening @hhovde,  
I am not really surprised at the difficulty of using the normally closed limit switches. It is indeed a little more complicated than those normally open.
Tomorrow morning, I will write the documentation page on the Wiki, I should have done it before publishing the version :smile:
@++;
Gauthier.

fra589, 2020-04-28:
Hi @hhovde,  
Here is the promised documentation:
In french: https://github.com/fra589/grbl-Mega-5X/wiki/Commutateurs-de-fin-de-course-et-prise-d'origine
Ou en Anglais: https://github.com/fra589/grbl-Mega-5X/wiki/Limit-switches-and-homing
Happy reading, tell me if something needs to be completed...
@++;  
Gauthier.

hhovde, 2020-04-30:
Thanks for your great efforts @fra589 

The new documentation gave valuable insight into the firmware, but I still can't get it to work as expected.
I've tried a number of different settings over the last couple of days, without success. I expect It might be a corner case of some sort.
However, I'm in the process of rebuilding and rewiring the entire machine, so I had to pause my efforts. As soon as I've got the machine back together I'll continue looking into the matter (probably sometime this weekend).

/H

Jagerk, 2021-09-17:
@hhovde Were you able to do more testing and/or find a solution to this issue? I too am having a similar issue to this and wanted to see if there was a configuration to solve this issue.
Just a little about my issue. When homing a dual motor (with dual endstops) one of the motors will initially move a little but then it will stop, the other motor keeps moving and causes the axis to twist.

RaphaelDives, 2021-09-17:
Hey,
had a similar issue, just something to think about - wiring of limit switches. I ran both motor (24v) and limit (5v) switch in one (long) cable. I think I got random values for my limit switches, but the behaviour was the same. One of the motors would eventually stop and twist the axis. (Testing Switches was always fine!)
Now my limit switches work with 24v and Relais, which completely solved the issue on my side… 😉
Cheers Raphael

Jagerk, 2021-09-17:
Hmm that is something to think about. I have the wires running in a metal tube. I'll go ahead and separate them out and see if that solves anything. 
Not really sure what you did with the relays and the limit switch. Could you show an example picture of this wired up? 
Thanks for the advice.

Jagerk, 2021-09-20:
@RaphaelDives tried the separating of the wires and it did not work for me. Will keep my eye on this post to see if any more potential resolutions come up. 
BTW if there are any test people want to run I’m totally up for being a guinea pig.

fra589, 2021-09-21:
Hi @Jagerk,  
  
Perhaps you can try to remap your limit switches to another pin of RAMPS/Arduino card...  
See https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h for how to do it.  
  
@++;  
Gauthier

Jagerk, 2021-09-22:
@fra589 I'll give that a try and see if it makes any difference. Will this affect how the motors work?
My main issue is that one of the motors does not move during homing. But I am willing to give anything a try.  
Thanks

fra589, 2021-09-22:
Hi @Jagerk,  
  
I suggested to remap the limit switch pin to see if there is an hardware problem on RAMP or Arduino about it. If you only remap the limit switch, this will not affect the motor work.  
  
You can also post your axis and homing definition in your config.h file and the result of the Grbl $I and $$ command, so I can see if your config seems correct.  
  
@++;  
Gauthier.

Jagerk, 2021-11-03:
@fra589 Sorry it has taken so long to get back to this, have had some other projects distracting me.
So I took your advice and remapped the limit switches and it did not make any difference on homing. I tried to enable some other features like "#define LIMITS_TWO_SWITCHES_ON_AXES", and also adjusted how I homed and it all did not do anything different.
Below is a branch I created where I changed in the config, cpu_map and my grbl $$. Take a look and see if you notice anything I should adjust. JagEEPROM is where the $$ settings are.

https://github.com/Jagerk/grbl-Mega-5X/blob/jagEdits/grbl/config.h
https://github.com/Jagerk/grbl-Mega-5X/blob/jagEdits/grbl/cpu_map.h
https://github.com/Jagerk/grbl-Mega-5X/blob/jagEdits/JagEEPROM

fra589, 2021-11-04:
Hi @Jagerk,  
  
Warning ! Your config.h and your cpu_map.h are not the one furnished with the 1.2d version... Some definitions are probably missing in your file for the 1.2d version...

I analyzed your configuration and here are my remarks:  

First, you don't need to enable the feature `#define LIMITS_TWO_SWITCHES_ON_AXES`. This is only needed if you have 2 limits switch wired in parallel on the same Arduino's pin. Leavi it commented. 
  
In config.h, you need to define:  
```
#define N_AXIS 5            // Number of axes (3 to 6)
#define N_AXIS_LINEAR 5     // Number of linears axis
```
As your cloned axis are also linear axis.
Reading your config.h, I understand you have:
- X axis not cloned on axis number 1,
- Y axis cloned on axis numbers 2 and 4,
- Z axis cloned on axis numbers 3 and 5.

For the homing cycle definition, you need only define the #if N_AXIS == 5 section (the other are ignored as you have defined N_AXIS to 5 :-) )
```
#elif N_AXIS == 5 // 5 axis : homing
  #define HOMING_CYCLE_0 ((1<<AXIS_3)|(1<<AXIS_5)) // Home Z axis first to clear workspace.
  #define HOMING_CYCLE_1 ((1<<AXIS_2)|(1<<AXIS_4)) // home Y axis.
  #define HOMING_CYCLE_2 (1<<AXIS_1) // Home X axis.
```
I read in your config.h than you tryed to use `INVERT_MIN_LIMIT_PIN_MASK` and `INVERT_MAX_LIMIT_PIN_MASK`...
- I understand all your limits switches are normaly closed switch,
- I understand your limits switches are on the MIN side for both X and Y axes and on the MAX side for axis Z.
I am right ?
As all your limit switches are not on the same side and all are normaly closed, I advise you to do otherwise:
- Leave `INVERT_MIN_LIMIT_PIN_MASK` and `INVERT_MAX_LIMIT_PIN_MASK` commented (don't use them)
- use instead the Grbl parameter `$5=1` to have all limits switchs with normaly close logic. So you will need to short circuit to the ground all unused limits pins.
As X and Y limits ars on the MIN side (negative direction) and Z on the MAX side (positive direction), you also need to define the Homing direction invert mask $23 as this: `$23=11` (binary &b00001011).

@++;
Gauthier.

Jagerk, 2021-11-05:
@fra589 
Took all of your advice and applied it. At first after I applied the changes I did not see any difference and was disappointed, but then I thought why not try and disconnect the y axis (y and z motor wires are running in the same tube) and then the homing started working properly, both z motors moved at the same time. So I think between what you told me to do and removing potentially Y axis interference it solved the issue. 
Bad news is that the second Y axis stopped working after I plugged it back in. I think I may have shorted something and caused the Arduino to stop working. Swapped drivers and used a different motor and nothing drives it. And before I unplugged it was working. Will be ordering a new Mega.
Thank you for the help, I think the Y should be good but can reply back once I get a new Arduino.

closed 2022-06-11 by fra589