fra589/grbl-Mega-5X #107: Post-processor ?

Flying-Hammer, 2020-03-26:
Hello hello
My 4axis laser engraver is now fully functionnal thanx to @fra589 
Unfortunately I can't find a Fusion 360 post-processor able to work properly
I try to develop my own, but it is a little bit tricky for me...
An idea for help, please ?
@++
JCG from France

fra589, 2020-03-26:
Bonjour Jean-Claude,
La défense de la liberté et du partage de la connaissance étant ma plus forte motivation dans l'écriture et le partage de mes compétences informatiques, je ne saurais trop te conseiller d'utiliser des logiciels open-source sous licence libre pour ta CAO.  
L'excellent [FreeCAD](https://www.freecadweb.org/?lang=fr) fonctionne bien avec Grbl. J'en sais quelque chose, c'est moi l'auteur du post-processeur FreeCAD-Grbl 😃 
Quand aux post-processeurs Fusion360-Grbl, je ne serais pas d'une grande utilité, je n'utilise pas Fusion et je ne veux surtout pas l'utiliser ! La liste des raisons est trop longue à expliquer pour être évoquée ici.
M'enfin, une simple recherche de "Fusion 360 post-processor" sur G**gle m'a conduit (parmi plein d'autres résultats) ici : http://wiip.fr/node/8549
C'est probablement un bon point de départ pour faire ce dont tu as besoin.  
@++;  
Gauthier.

closed 2020-03-26 by fra589

henpemaz, 2020-03-31:
I do a lot of my projects in Fusion 360, I guess having cad and cam in one package is hard to beat.

For 3-axis machining I've been using the GRBL post that is available now from their Posts library (it got updated last year, the one that was there before was unusable). In fact I've tweaked it a little to get around some issues like clearing the workpiece (which it didn't do) when G28 is set not to be used.

Now for 4 (and maybe 5 in the future) axis machining, that same post won't work. I'm scared of trying random ones from their library. I did test the Carbide 3d one though, which is based on GRBL, but it also doesn't support >3 axis.

I might give it a go on trying to make the current GRBL post there compatible with 4 or 5 axis, but it looks a bit intimidating...