fra589/grbl-Mega-5X #108: Compiling question

Terixxx, 2020-03-26:
Hello, 

I have chinese CronosMaker 4 axis control board. When i flash the board with pre-builded grbl-Mega-5X-v1.1l.20190605.hex everything work except limiting switches. The CPU_pin mapping on the board is the  same as in the source code. 
I try to copmile latest release but when i flash firmware the board is not working at all. What i miss in the configuration and compilation? Please help me to compile it with enabled limited switches.

fra589, 2020-03-26:
Hi @Terixxx,  
What system do you use for compiling and to flash Grbl on your card ?
Under Windows, the recommended method is to use Arduino software and it's described here : https://github.com/gnea/grbl/wiki/Compiling-Grbl.
The v1.1l version of grbl-Mega-5X is out to date, but I no longer wish to provide the compiled versions, since they are too limited and specific for only one config.
I therefore advise you to download the latest version and use the procedure cited above for your needs.
@++;
Gauthier.

Terixxx, 2020-03-26:
Thank you very much for your fast e correct assistance. following the instructions i was able to campile and flash the firmware successfully for my board. All functionality is working as expected including hardware limiting switches. Thank you very much.

closed 2020-03-26 by Terixxx