fra589/grbl-Mega-5X #109: pinout arduino Mega 2560

msaarongonzalez, 2020-03-28:
Hi  thanks for taking time to help me. Well i am using  tb6600 drivers  working on a Arduino mega 2560, but i have a problem, i have upload the hex file, ( both version 20190605 and 20180712) 
[image]
[image]

but when i try to probe the motors send me and error " Alarm 2 , reset to continue" i have try the normal version of grbl and they work well, so it is not the conextion. the  problem is that i need to work with 4 axis  and 1 slave on the y axis, in total 5 motor and 4 axis, What would be the solution ? 

PLease what are the pinout  for rotary axis  (A,B) on arduino mega, i have been looking buti cant find then with out using the ramp 1.4 shield.
I will be so much gratefull i some one can help me

fra589, 2020-03-30:
Hi @msaarongonzalez,  

> PLease what are the pinout for rotary axis (A,B) on arduino mega, i have been looking buti cant find then with out using the ramp 1.4 shield.

You can see how the pin mapping is build in the cpu_map.h file [in the wiki](https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h).  
And then, reading the cpu_map.h will give you the response.

>  I will be so much gratefull i some one can help me

Refer to https://github.com/fra589/grbl-Mega-5X/blob/edge/doc/csv/alarm_codes_en_US.csv:

2 | Soft limit | Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked.
-- | -- | --

That means than the movement you ask is out the limit space of your CNC machin.
After homing, Grbl will set by default the entire machine space into negative space, as is typical for professional CNC machines, regardless of where the limit switches are located.
So, just after homing and without defining other coordinate system, the only move you can do is towards negative coordinates.
I advise you to read the excellent documentation of Linux CNC about coordonate system:  
http://linuxcnc.org/docs/2.7/html/gcode/coordinates.html

@++;  
Gauthier.

msaarongonzalez, 2020-03-31:
Thank you so much, really useful information. I have one more question: it is posible to clone one axis? What's the proceeding

fra589, 2020-03-31:
Look in the previous issues.
Before asking a question, it is a good idea to check if this question has not already been answered.
A simple question to **[Google](https://www.google.com/search?q=grbl-mega-5x+clone+axis)** also often gives quick answers.

closed 2020-03-31 by fra589

atila120, 2021-01-24:
Hi, I am uploading the file on the 2560 board, but my stepper motors are not moving