fra589/grbl-Mega-5X #111: Homing works but hard limits don't

d01010101, 2020-03-30:
Hi, I use grbl-Mega-5X with RAMPS 1.6. I connected 6 limit switches via this board. It seems that the switches are ok: homing works, thus I assume that X+ Y+ Z+ are ok, pressing e.g. X- during initialisation shows "check limits" as expected. I have even set temporarily $23 to test if the inverted homing works, no problem. Soft limits are disabled $20=0, hard limits are enabled $21=1.

Despite all this, hard limits do not work. For example, pressing the same switch, which caused "check limits" during initialization, does not stop anything. The machine ignores its hard limits in any direction, including the homing directions.

I use a fairly recent shapshot from github. What can I test, do to solve the problem?

Config:
$0=10
$1=254
$2=0
$3=0
$4=0
$5=0
$6=0
$10=16
$11=0.020
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=0
$24=100.000
$25=600.000
$26=250
$27=5.000
$30=12000
$31=550
$32=0
$100=800.000
$101=800.000
$102=650.000
$103=8.889
$104=8.889
$110=1000.000
$111=1000.000
$112=900.000
$113=1440.000
$114=1440.000
$120=50.000
$121=50.000
$122=50.000
$123=50.000
$124=50.000
$130=300.000
$131=180.000
$132=100.000
$133=360.000
$134=180.000

fra589, 2020-03-30:
Hi @d01010101,  
Do you have the last version?
What give the $I output?
Hardware limits are implemented in RAMPS version only since the v1.1m version...
Have a look to older issues... For example here: https://github.com/fra589/grbl-Mega-5X/issues/15#issuecomment-565830405
@++;  
Gauthier.

d01010101, 2020-03-30:
Hi fra,
$I < [VER:1.1m.20191215:]
[AXS:5:XYZAB]
[OPT:VNMH,35,255,24]

The switches are recognized, as I said (e.g. homing is possible or "check limits") but otherwise, the machine does not stop when hitting one. No alarm etc.

d01010101, 2020-03-30:
Is there any way of displaying the status of the switches, as seen by the firmware?

fra589, 2020-03-30:
Hi @d01010101,  

The Status Report Query "?" give the switchs status in the "Pn" field like this:  
`<Sleep|WPos:0.000,0.000,0.000,0.000,0.000|Bf:35,255|FS:0,0|Pn:XYZABC>`
It show the letter axis only when it trigered.
  
OK, you have the good version.
As I write it before in the [comment I gived you the link](https://github.com/fra589/grbl-Mega-5X/issues/15#issuecomment-565830405), You need to enable the feature in cpu_map.h (line 321) by removing the comment (`//`) before the sentence:
`#define ENABLE_RAMPS_HW_LIMITS`
Do you have made this?
@++;  
Gauthier.

d01010101, 2020-03-31:
No, I did not about this. Now it works, thank you.

closed 2020-03-31 by d01010101

wwwlxs, 2020-07-14:
My grbl has the same problem, sometimes works but sometimes not, very unstable, it is often to be reset.

wwwlxs, 2020-07-14:
And It often comes with “error 9”

fra589, 2020-07-15:
@wwwlxs asked:
> And It often comes with “error 9”

Already replied here :
https://github.com/fra589/grbl-Mega-5X/issues/1#issuecomment-382366866

And please don't squat closed topics that have already been answered. Open your own!

@++;  
Gauthier.