fra589/grbl-Mega-5X #112: Different pwm pin for laser and spindle

MarkXYZ-Mtl, 2020-04-04:
Hi, 
Like the title says, is it possible to have 2 different pin for spindle and laser triggered by the $32 variable, it can be a relay output too.
I want to have both laser and spindle mounted on the machine.
I had a look at the code but without success.

Thanks

fra589, 2020-04-04:
Hi @ MarkXYZ-Mtl,
It's a good idea...
Using $ 32 is a possibility, but not the only one ...
In fact, the idea is to have 2 PWM outputs: One on D8 in 0-12V and the other on another port in 0-5V.
We can therefore also imagine using new commands such as M3.1, M4.1 and M5.1 to control the second output.
I add this to my TODO list. 😃 
@++;
Gauthier.

MarkXYZ-Mtl, 2020-04-06:
Hi,

The use of a special M-code can be troublesome for cam software without editing the post processor. 

Can be an option to choose either special M code or $32? 

And it would be nice if the $32=1 can activate a relay for the main laser power for safety reasons. 

Thanks for your interest!

fra589, 2020-04-07:
Hi,  
As I writed:  
> Using $ 32 is a possibility, but not the only one ...  
  
I analyze a need, not a solution...  
  
The need is to have more than one PWM output to drive the tool with a choice of 5V or 12V PWM without the need of recompiling the firmware.  
   
$32 is a candidate solution, but not the only one for doing this.  
  
The original need can give other oportunities to have other benefit such having more than one tool using both 12V and 5V PWM output in the same GCode program, in this case, $32 will not be candidate since it's not a GCode order.  
  
The final implementation probably offer more than one solution with more than one compil option.  

@++;  
Gauthier.

martinjurekcz, 2020-05-02:
Hello guys, that would be really amazing. Does the arduino board have any PWM pins left for this? Im designing laser engraving machines. I already designed new prototype and CAM software which would use multiple laser beams to cut down production time. However I can’t find control firmware which would be able to control multiple PWMs as an output for laser modules. Can I help to make it happen? I dont know grbl so much, but few advices on where to start would really help :)

zalexzperez, 2021-12-06:
Is this feature request what I need if I want to control the speed of a fan connected to an MKS GEN L board? 

It's for a foam cutting machine. The hotwire is currently controlled with the heated bed output (D8 pin) which is PWM regulated. But the extra pins (hotends, fan...) are either fully on or off.

fra589, 2022-06-24:
Hi @MarkXYZ-Mtl, @martinjurekcz, @zalexzperez, @RaphaelDives,  
  
I just pushed the new 1.2g.20220620 grbl-Mega-5X version with 2 new main features:  
- Possibility to use 2 distinct output pin for Spindle and for Laser. The distinction is made by the use of $32 "laser mode" parameter.
  - If $32=0, the PWM spindle output is going to the spindle pin,
  - if $32=1, the output PWM of M3/M4Sxxx goes to the laser output pin.  

  2 new settings have been added to configure the min / max output values when grbl is in laser mode:
  - `$33`: Maximum laser value (depend of Laser unit) set the maximum laser value (set PWM to 100% duty cycle).
  - `$34`: Minimum laser value (depend of Laser unit) set the minimum laser value (Set PWM to 0.4% or lowest duty cycle).

  For compatibility raisons, this functionality is disables by default. To use it, you need to enable `#define SEPARATE_SPINDLE_LASER_PIN` line in config.h (line 146).  

- Supplemental optional analog PWM output:  
  grbl-Mega-5X can now drive a new analog output with the use of GCode commands M67 / M68:
  - `M67Q<value>` set the analog output to the desired value synchronized with motion (M67Q0 turn off the analog output in synchronisation with motion),
  - `M68Q<value>` immediately set the analog output to the desired value (M68Q0 immediately turn off the analog output).

  2 new settings have been added to configure the min / max output of analog output:
  - `$35`: Maximum output value (Volts or other units) set the maximum output value (set PWM to 100% duty cycle).
  - `$36`: Minimum output value (Volts or other units) set the minimum output value (sets PWM to 0.4% or lowest duty cycle).

By default, $35 and $36 are defined to be unit value given in volts (depending of the RAMPS output pin 12v for D8/D9 or 5v for D6). Then, if you send M68Q3, the output will immediately be approximatively 3v (depending of your electronic precision).

Since I don't have any laser or special device to test those functionality in the real world (I only verified the outputs with multimeter) please, test it and report if it work for you or if you have any problem with this version.
Thanks in advance for your tests reports.  

@++;
Gauthier.

martinjurekcz, 2022-06-28:
Hi @fra589 that sounds amazing ! :)) I will test it during july and report the performance. Can't wait for it. Thanks again, you are our hero 💯 Will report my tests soon,
have a great day,
Martin

RaphaelDives, 2022-06-28:
Same here!
Thank you so much!
I will test it on my MPCNC asap and report back! 
(Although I need to do some hardware changes.)
Cheers 
Raphael

zalexzperez, 2022-09-11:
Hi @fra589 

I tested the latest grbl-mega-5x version with bad results :(

Machine working fine on a MKS Gen L board. It is able to  jog, home and D8, D9, D10 outputs are properly enabled.

However, enabling USE_OUTPUT_PWM in config.h makes the machine break. It doesn't jog nor is it able to enable any outputs. Homing works, though...

This is what the console throws when jogging:

> client $J=G21G91 X1250 F3000
> error:28 (Invalid gcode ID:28)
> client \x85

This is what the console shows when trying to enable some outputs:

> feeder M3 S21570
> error:28 (Invalid gcode ID:28)
> feeder M7
> error:28 (Invalid gcode ID:28)
> feeder M8
> error:28 (Invalid gcode ID:28)
> feeder M9
> error:28 (Invalid gcode ID:28)

In config.h I literally ONLY changed the following:
-Axis definition (4-axis)
-Disabled HOMING_INIT_LOCK
-Different homing sequence to only home the X-axes 

 I didn't touch anything else in config.h file or anywhere else! I'm attaching my [config.zip](https://github.com/fra589/grbl-Mega-5X/files/9542447/config.zip) here, you can search for "//Changed" to quickly see my changes.  

And I did perform a $RST=* after flashing the new firmware and also disabled the soft limits. Tried twice with the same result.

fra589, 2022-09-14:
Hi @zalexzperez,  
  
Thanks for reporting...  
Reminds me of the typo bug in gcode.c that I fixed shortly after...  
Can you post the output of the $I command so that I can check the version and compilation settings of your Grbl?  
  
@++;   
Gauthier.

zalexzperez, 2022-09-14:
Hi @fra589 
This is the output:

>client $I
>[VER:1.2g.20220620:]
>[AXS:4:XY]
>[OPT:VNMGHQL,35,255,80]
>ok

fra589, 2022-09-17:
Hi @zalexzperez,  
  
It was a new bug!  
I found and corrected it. 
The new version 2.1g build 20220917 have been pushed few minutes ago.

Thanks for testing !

@++;
Gauthier.

zalexzperez, 2022-09-17:
Hi @fra589  

Thanks for fixing that bug. 
It works now, look: https://youtu.be/LNCHpRLxbcY

However, my plans were to control two 12v fans with the extra PWM output. Unfortunately, they all make a horrible noise when powered from anything else other than constant 12V.

I suffered from this noise in my 3D printing running Marlin for a while, until I found there's an option to enable software PWM on the fan output pin, which is exactly D9.

On Marlin's Configuration.h file, there's a FAN_SOFT_PWM option that I had to enable, along with SOFT_PWM_SCALE 
 which halves control resolution and SOFT_PWM_DITHER to compensate for that resolution loss.
Once enabled, the fans almost make no whining sound when run at lower speeds.

Would it be possible to port this feature? 

Regards, 
Alejandro

Host1992, 2022-10-12:
does anyone tested the "SEPARATE_SPINDLE_LASER_PIN" with succes? 
im stuck , my spindle relay pin is on D4, my laser pwm is on D6, (as set in my config.h) I tried everything but no pwm come out from D6

fra589, 2022-10-20:
Hi @Host1992,  
  
Spindle on D4 is not the normal pinout output for spindle... It looks like you modified the cpu_map.h to send the spindle output to D4.  
Warning !!! D4 use the Timer0 for PWM which conflicts with steppers management ! You can't use it for spindle output.
  
Please, post yours config.h, cpu_map.h, `$I` and `$$` outputs for more investigations of your problem.  
  
@++;  
Gauthier.

fra589, 2022-11-21:
No news from near a month... Issue closed.

closed 2022-11-21 by fra589