fra589/grbl-Mega-5X #114: Homing G28 and

kikirikiki, 2020-04-13:
Hi, my machine is working perfect but when I send a g-code with a raspberry pi, I don´t know what I have to write in the G-Code to unlock the grbl. With Windows and Universal Gcode Sender works with #H but without the GUI of Universal Gcode Sender, I can not start it. What I have to write in the gcode to unlock the machine?
Another thing that I would like to change is the behavior with G28. The machine goes to the start point  after homing but it do not touch the end stop. What I have to write in the gcode to do a proper homing?
Thanks a lot!

fra589, 2020-04-14:
https://github.com/fra589/grbl-Mega-5X/wiki
https://github.com/gnea/grbl/wiki
And for more general infos:
http://linuxcnc.org/docs/html/index.html

kikirikiki, 2020-04-14:
Thank you very much. I´ve read this documents and I can not unlock the GRBL without the GUI in Windows. I can not send a file.gcode direct from my Raspberry Pi automatically without clicking the $H buttom or send the $H command in this GUI. I can not send a "Homing.gcode" with an $H inside, GRBL doesn't recognize this command.
Thanks!

fra589, 2020-04-15:
It seems you don't have read the full documentation... (many links on the right inside the [original Grbl wiki page](https://github.com/gnea/grbl/wiki)

> direct from my Raspberry Pi

??? You don't use a [Grbl interface program](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface) on your Pi?
It is magic! how do you do send your GCode file without an interface program?

kikirikiki, 2020-04-16:
Sorry, but I have read this documents and I tried to send a $H command with a gcode file through Windows GUI and I get the same error 
[image].

How could I disable the file streaming lock without sending the $X command in Universal Gcode Sender?
Thanks

fra589, 2020-04-16:
> Sorry, but I have read this documents

Donc, comme tu as pu le lire la réponse à ta question est ici : https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#22---homing-cycle-boolean:

> (...)
> Also, one more thing to note, when homing is enabled. Grbl will lock out all G-code commands until you perform a homing cycle. Meaning no axes motions, unless the lock is disabled ($X) but more on that later. Most, if not all CNC controllers, do something similar, as it is mostly a safety feature to prevent users from making a positioning mistake, which is very easy to do and be saddened when a mistake ruins a part. If you find this annoying or find any weird bugs, please let us know and we'll try to work on it so everyone is happy. :)
> 
> NOTE: Check out config.h for more homing options for advanced users. You can disable the homing lockout at startup, configure which axes move first during a homing cycle and in what order, and more.
> 

Your problem is not a Grbl or file streaming problem.

It is your Windows interface tool that is programmed not to start streaming while Grbl is in Alarm mode.

This alarm mode is normal and desirable for safety reasons when the power is turned on. I do not recommend disabling it, but as it is written in the documentation, it is possible... Read the documentation.

closed 2020-04-23 by fra589