fra589/grbl-Mega-5X #115: A-axis blocked

crkie, 2020-04-16:
Hello,
If the A-axis (4 non-linear) exceeds Max Travel, it remains blocked. A reset does not help, ALARM 2 always comes up. It only runs when $ 20 = 0. Can't I use soft limits for rotary axes?

Best regards
Rudi

fra589, 2020-04-18:
Hello [Rudi](@crkie),  
For me, it's the same behavior than linears axis. The axis remain in Alarm mode when you ask a travel that exceed the max travel value. So, you need to unlock the Alarm mode.
The Grbl unlock alarm is $X, not ^X (<Ctrl>+X) wich is the soft reset command, and then, if you only issue the ^X soft reset, the Alarm mode is not cleared.
@++;
Gauthier.

crkie, 2020-04-18:
"it's the same behavior than linears axis" 
Of course, without an offset, I can only work in the negative range. I had tried positive values.
Thank you
Besides, can I somehow switch off the soft limits for the rotary axes and leave them at linear axes?

fra589, 2020-04-18:
Setting the max travel parameter of an axis to 0 will disable soft limit to this axis
(parameter $130 to $135 depending of axis you want)
@++;
Gauthier.

crkie, 2020-04-18:
Great, now it works the way I wanted it to work

merci
Rudi

closed 2020-04-18 by crkie