fra589/grbl-Mega-5X #116: Bug in gcode.c causes Jog issue (using Universal Gcode Sender).

mattknudstrup, 2020-04-19:
Line 1193, parenthesis misplaced:
    if (command_dwords & ~(bit(MODAL_GROUP_G3) | bit(MODAL_GROUP_G6 | bit(MODAL_GROUP_G0))) ) { FAIL(STATUS_INVALID_JOG_COMMAND) };
...should be...
    if (command_dwords & ~(bit(MODAL_GROUP_G3) | bit(MODAL_GROUP_G6) | bit(MODAL_GROUP_G0)) ) { FAIL(STATUS_INVALID_JOG_COMMAND) };

fra589, 2020-04-19:
Hi @mattknudstrup,  
Oops! you must be right. It's a typo, I will correct it ASAP...
Thanks for reporting,
@++;
Gauthier.

fra589, 2020-04-20:
Hi @mattknudstrup,  
I just pushed a commit correcting this issue, it's was too easy since you reported the bug location :smile:.
Thanks to test and report if OK.
@++;
Gauthier.

fra589, 2020-04-24:
Hello @mattknudstrup,  
Doing this test is not very important now, I trust that it was only a typo. I will check by the next release.
Thanks to make open source better :-)
@++;
Gauthier.

closed 2020-04-24 by fra589