fra589/grbl-Mega-5X #117: Rotation axis problem.

Pavshi, 2020-04-22:
Hi @fra589 
On my cnc the fourth axis is a tangential knife.
when i send the command G2/3 like: 
G02 X46.336814 Y25.635248 Z-1.000000 I-8.700505 J3.892172 A-2.39463684581
the knife does not turn while moving in an arc,
he turns after moving, at the end of the arc :-((
Can you help me with this problem?
I'm trying to figure out gcode.c but so far without success.

I'd like to solve it with your help.

fra589, 2020-04-22:
Hi @Pavshi,   

Circular interpolation work like defined in the [LinuxCNC documentation](http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g2-g3)...
But in this documentation, only a third axis is used for helix movement. So, I don't study other special cases like the one you need now...
I think perhaps it will be possible, but not so easy :-$
Then, it only will be a time linear interpolation between the start position and the final position (at N% of circular movement, the 4th axis will have the same N% of his total movement).
So, your knife will not be exactly tangent to the circle...
Perhaps it will be better for you to make those movements with many linear movements interpolation.

@++;
Gauthier.

By the way, dont forget to tell your exact version and config (output of $I and $$) when you ask for a new issue...

Pavshi, 2020-04-23:
Hi, Gauthier!
thank you very much for such a quick reply :-)
Here is the version I last used:
$I
[VER:1.1h Atmega 2560 6-axis + Sync/MPG.20180311:]
[OPT:VNM]
ok

Now I uploaded your latest version to Arduino, but have some problems with pins.
I don't use Ramps. And I'm a little confused. 
Only x,y,z  in top of cpu_map.h.

$$
$0=10
$1=255
$2=0
$3=1
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=538.182
$101=1613.115
$102=133.124
$103=1152.377
$104=250.000
$105=250.000
$110=800.000
$111=800.000
$112=800.000
$113=1500.000
$114=500.000
$115=500.000
$120=50.000
$121=50.000
$122=50.000
$123=50.000
$124=10.000
$125=10.000
$130=200.000
$131=200.000
$132=100.000
$133=200.000
$134=200.000
$135=200.000
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok

fra589, 2020-04-24:
Hi @Pavshi,  
Your `$I` output don't come from my own grbl-Mega-5X version!?  
I can't debug other versions since grbl-Mega-5X already take me a lot of time!

For my own version:
grbl-Mega-5X support more than 3 axis only with the CPU_MAP_2560_RAMPS_BOARD definition.
So, even if you don't use the RAMPS board, you need to define in config.h:
```
#define DEFAULTS_RAMPS_BOARD
#define CPU_MAP_2560_RAMPS_BOARD
```
and use the section after the sentence `#ifdef CPU_MAP_2560_RAMPS_BOARD`, line 136 in cpu_map.h.

There, you will find where to plug your hardware or to change the definitions to conform to your need, see the wiki page: https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h  

@++;  
Gauthier.

Pavshi, 2020-04-24:
Bonsoir, Gauthier!
I finally figured out the ports and pins :-)
And voilà :-)))
It's working !!!
Your version really do it (4th axis rotates by G2/3). 
I’m still testing it, but what I see is very good.
Now I'm happy, I can say good buy to Windows and Mach3 :-)))
Thank you so much for your work.
$I
[VER:1.1m.20200420:]
[AXS:5:XYZAB]
[OPT:VNMH,35,255,0]
ok
:1st_place_medal:

closed 2020-04-24 by Pavshi