fra589/grbl-Mega-5X #118: 3 axis switchable to 4 axis

supernono-fr, 2020-04-25:
bonjour Gauthier

je cherche un moyen d'avoir un switch en facade de ma cnc pour pouvoir mettre en service ou hors service le 4ème axe 
ma cnc est définit comme 5 axes ,4 linéaires et Y cloné.   XYZAY'

après divers test je me suis aperçu que si je déconnecte le 4ème axe (moteur et fin de course) le homing plante, ce qui semble normal vu que le fdc est déconnecté  
j'ai bien une idée pour le faire mais je ne connais pas trop bien le langage C utiliser par grbl:
en mettant une condition dans les définitions du homing dans le config.h 
``
#ifdef DEFAULTS_RAMPS_BOARD
  #elif N_AXIS == 5 // 5 axis : homing
   # if switch == 1
      #define HOMING_CYCLE_0 (1<<AXIS_3) // Home Z axis first to clear workspace.
      #define HOMING_CYCLE_1 (1<<AXIS_1) // Home X axis
      #define HOMING_CYCLE_2 (1<<AXIS_2)|(1<<AXIS_5)// Home Y & Y' axis 
     #define HOMING_CYCLE_3 (1<<AXIS_4) // Home 4th axis (A)
  # else switch == 0
     #define HOMING_CYCLE_0 (1<<AXIS_3) // Home Z axis first to clear workspace.
     #define HOMING_CYCLE_1 (1<<AXIS_1) // Home X axis
     #define HOMING_CYCLE_2 (1<<AXIS_2)|(1<<AXIS_5)// Home Y & Y' axis 
     //#define HOMING_CYCLE_3 (1<<AXIS_4) // Home 4th axis (A) 
  #endif  
#endif 
``
mon problème c'est que je ne sait pas ou et comment définir une entrée pour mon switch dans grbl saurait tu m'aider dans ma démarche ?
 
j' utilise une carte MKS GenL (carte dérivée de la ramps 1.4)
j'ai créer une "définition" dans le fichier cpu_map.h pour cette carte avec la possibilité d'avoir les sorties coolant mist ,coolant flood et spindle pwm soit sur les mosfet (heater E0 heater E1 hotbed) soit sur des sorties 5v (D52 D54 servo4) si cela t'intéresse pour l'intégrer a ton travail je peux te le communiquer avec plaisir 

Arnaud

fra589, 2020-04-26:
Bonjour Arnaud,
Ce que tu veux faire n'est pas possible.
En effet, `#define` n'est pas une commande du langage C, mais une commande du pré-processeur du compilateur. Ce qui veux dire que cette commande n'est exécutée qu'à la compilation du programme lors de la constitution du code exécutable et non lors de chaque exécution du programme final.
Quand on écrit `#define TOTO 5`, on définit une constante et le compilateur remplace TOTO par 5 partout ou il est utilité dans le code source au moment de la compilation.
Ce que tu veux faire n'est pas possible, mais tu peux contourner le problème en simulant le homing du 4ème.
Soit tu garde branché le switch de fin de course du 4ème axe, soit tu en connecte un autre en parallèle.
Au moment ou Grbl effectue le homing, tu laisse faire les 3 premiers axes normalement, et lorsqu'il arrive au 4ème, tu actionnes manuellement le switch 2 fois, comme si le contact était actionné par l'axe, puis l'axe se retire de quelques millimètres, puis, reviens à vitesse lente pour la mesure précise.
C'est comme ça que je fais quand je suis amené à tester des configurations sans le matériel complet.
@++;
Gauthier.

supernono-fr, 2020-04-26:
bonjour Gauthier 

merci de ta réponse rapide, je vais donc mettre un faux fin de course pour simuler le homing du 4ème axe

autre question pour le fin de course A j'ai prévu un disque avec une fente étroite solidaire de l'arbre de l'axe et un fin de course avec une fourche optique 
si on active les soft limit on est obligé de définir une course maxi pour tous les axes 
du coup pour l'axe A il n'est pas possible d'avoir une rotation illimité (exemple sans intérêt réel mais pour illustrer :tailler une tige filetée a partir d'un rond avec une fraise à 60°) ?
 
Arnaud

fra589, 2020-04-26:
Bonjour,
Cette question a déjà été abordée et obtenu réponse, la dernière fois il n'y a pas longtemps : https://github.com/fra589/grbl-Mega-5X/issues/115#issuecomment-615899061
C'est bien de chercher un peu avant de demander...
@++;
Gauthier.

closed 2020-04-26 by fra589

supernono-fr, 2020-04-26:
désolé je ne l'avais pas vu 

merci pour tout ton travail