fra589/grbl-Mega-5X #119: RAMPS board code Vs Generic

revyradness, 2020-04-28:
Hi,
I'm curious what the difference is between the generic code and RAMPS Board specific code throughout the project?

As I want to use a cloned axis (clone A to Y), but that is currently only supported using the RAMPS board (which I don't have) and not the generic one, I just updated the pins in the RAMPS_BOARD section of the pin map to match a Generic Arduino Mega 2560 and it works fine. 

If this is possible, why is there a need for all the conditional code?

Thanks!

fra589, 2020-04-29:
Hi @revyradness,  
  
## Simple answer: This is historic :grin:  

## Little more complete answer: :thinking:  
    
The original porting of Grbl between the Arduino Uno card and the Mega largely retained the limitations of the original card.  
  
The main imitation being that the pins that wich conduct the drivers of the stepper motors use the same processor port for all of the axes.
For the limit switches, only a few pins are compatible with the triggering of processor interruptions, necessary for managing the hardware limits.  
  
These constraints being incompatible with the use of the RAMPS card, a lot of additional and strongly different code was necessary to adapt the use of RAMPS. This resulted in a slight drop in performance as well as the inability to use the hardware limits. This is why the 2 versions have been kept.  
  
This was the status of the grbl-Mega projectwhen I started to work on it.  
  
When I started the grbl-Mega-5X project, which consisted of adding the additional axes then allowing the cloning functions and other things, I naturally used the version with RAMPS which directly has the outputs for the 5 motors. My firsts tests was made on my Prusa 3D printer which was equiped with Mega + RAMPS.  
So I added and modified the code in the RAMPS_BOARD section while keeping the rest of the code intact, which ensured that it stay possible to use the program as before.  
The adaptation of the original section seemed to me too complicated to be able to respect the constraints described above, so I did not modify it.  
  
My primary intention was to merge my additions with the main grbl-Mega project to keep only one version. And to participate thereafter in the maintenance of the main branch of Grbl.  
Sonny having declined my proposal, I continued on my own the development and maintenance of grbl-Mega-5X.  
  
I actually intend to clean up the code of all the elements that are no longer used...  
This is part of my TODO list...  
When will I do it?  
Probably soon :blush:  
  
@++;  
Gauthier.

RaphaelDives, 2020-04-29:
👍 
I just want to take the opportunity and leave a BIG Thank You @fra589!! :)
Cheers
Raphael

revyradness, 2020-04-29:
Thanks so much for the information!

I came across 5X because I was looking for a build that supported a cloned axis on the Mega as Gnea/Grbl-Mega is a couple of releases behind Gnea/Grbl-Mega so doesn't have it. 

I hope my initial question didn't come across incorrectly. It's only that I'm a completely novice coder and was simply curious!

Thanks very much for this project. Much appreciated.

fra589, 2020-04-29:
Hi @revyradness,  

> I hope my initial question didn't come across incorrectly. It's only that I'm a completely novice coder and was simply curious!

You are welcome :slightly_smiling_face:, there are no bad questions, except those that have already been asked and have already been answered.
Often, the most difficult thing in software development is to take into account the past so as not to destroy it when building the future...

@++;
Gauthier.

closed 2022-06-26 by fra589