fra589/grbl-Mega-5X #121: CoreXY on 4 axis

oilytin, 2020-05-10:
would it be possible to add coreXY kinematics on two sets of axis XY and UZ?
I've gone and built a 4 axis hotwire before realising the limitations of the firmware.
I have started looking at the code and fumbling my way through it.
would you be able to give me your opinion(and any advice).
I have only picked XYUZ as the gcode sender i was thinking of using seems to be stuck with these definitions [RC groups thread](https://www.rcgroups.com/forums/showthread.php?2915801-4-Axis-Hot-Wire-CNC-%28Arduino-Ramps1-4%29-Complete-Solution). 
A 4 axis hotwire is generally arranged as per the image below. For my applications the axis would need to move independently in order to create tapered parts.
[image]

fra589, 2020-05-10:
Hi @oilytin,  

Since your hardware configuration doesn't have Z axis, the standard CNC axis for this configuration is XY and UV cf. [this documentation about standard axis definition](http://linuxcnc.org/docs/html/gcode/machining-center.html#_axes): 
> A foam-cutting machine may have XYUV axes..

The axis definitions in thn RC groups thread is wrong, due to the the constraint of the original axis name definition in the Letartare version, and due to the fact that the author of the machine did not have the courage to rename them.
This constraint is not present in grbl-Mega-5X as each axis can have his name defined in the config.h file. So, I considere than you should use the standard definition of axis for your need : X, Y and U,V.

So, you need to add the code for the COREUV mode...

First, define a new compilation option "COREUV" in the **config.h** file like the "COREXY" one:
```
#define COREXY
#define COREUV

```
In **nuts_bolds.h**, define the COREUV motors like this:
```
#ifdef COREXY
 #define A_MOTOR AXIS_1 // Must be AXIS_1 (X)
 #define B_MOTOR AXIS_2 // Must be AXIS_2 (Y)
#endif
#ifdef COREUV
 #define C_MOTOR AXIS_3
 #define D_MOTOR AXIS_4
#endif
```

Then, search for all usages of the COREXY option in all files to duplicate the considered code and adapt it. You will find it in those files: **limits.c**, **planner.c**, **report.c**, **system.c** and **system.h**.

It shouldn't be too complicated to do.

Good luck,
@++;
Gauthier.

oilytin, 2020-05-10:
Thanks for the pointers Gauthier,
definitely a steep learning curve for me but looking forward to the challenge.

oilytin, 2020-05-19:
Hi Gauthier, have managed to add the required code an it appears to be working as expected. it is jogging as expected but haven't sent any gcode yet.
have also found improved software to control it aswell
https://github.com/mstrens/Hot_wire
https://github.com/oilytin/grbl-Mega-5X/tree/oilytin-patch-1

JPGDubois, 2021-02-15:
Hi @oilytin , I'm also trying to make a similar machine. Could you maybe share the modified code? That would be really helpful as I wasn't able to modify the code myself.

oilytin, 2021-02-15:
I made a fork here
https://github.com/oilytin/grbl-Mega-5X/tree/oilytin-patch-1
im not the best at git hub though. cant remember how up to date it is. it looks to have the 4 axis corexy implemented though.

apart from the extra corexy calculations needed there are also some homing issues (if your planning on using homing?)
see this thread
https://github.com/fra589/grbl-Mega-5X/issues/124