fra589/grbl-Mega-5X #123: Using D8 Spindle PWM with 24V Ramps

MCU3291, 2020-05-22:
Hi,
I would like to use grbl 5x in combination with a Ramps 1.4 board and an Arduino Mega2560 to operate my CNC milling machine with it. The ramps was converted to use with 24V (i removed the diode to the arduino board and changed the fuses). Now I would like to use the PWM pin D8 to control the spindle. 
What do I have to set as PWM Max / MIN to get a range of 1-10V from pin D8 with the supply voltage of 24V? I've tried different settings but non of them worked how i would expected it :/.

fra589, 2020-05-23:
Hi @MCU3291,  
Have you seen this?: https://github.com/fra589/grbl-Mega-5X/blob/edge/doc/script/fit_nonlinear_spindle.py
This script permit to calculate all PWM parameters, and it's comments explain how PWM spindle work in Grbl and how it can be adjusted.  
@++;  
Gauthier.

MCU3291, 2020-05-23:
I'll take a look, thanks!

MCU3291, 2020-05-23:
So i've taken a look at this, but this is not what i was looking for. There must be a more simple way to calculate the needed PWM values as they are simple represent the duty cycle settings.

So it would be awesome if you could answer me the following questions:

-what resolution has the timer used for D8? 8, 10 or 16bit?
-what frequency do you set up as default? the frequency should be a fix value,  one should only need to alter the duty cycle.
- is there a way to invert the rpm at min / max pwm to max /min pwm?

Thanks in advance!

fra589, 2020-05-25:
Hi @MCU3291,  
  
The script which I put the link below allows to calculate all the parameters of the PWM output, including in case of non-linearity when you enable piecewise linear model of the spindle PWM/speed output (cf. config.h lines 690-707).  
  
The PWM output is calculated to obtain the desired spindle speed with the with input of $30 and $31 settings, respectively rpm_max and rpm_min settings in agreement with the SPINDLE_PWM_RANGE (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE) and other values defined in the cpu_map.h file (line 360 to 384). All the computations and parameters for spindle PWM output are made in the spindle_control.c file.  
  
As I can read in the comments, the spindle PWM output use a 16-bit Fast PWM 1.9 kHz frequency at 1/8 prescaler.  
  
I didn't write this part of the Grbl code myself and I don't understand what you exactly whant. So I can't help you and doing for you that you can do yourself: Read the spindle_control.c file and analyse it with the lighting of the [ATmega2560 datasheet](https://ww1.microchip.com/downloads/en/devicedoc/atmel-2549-8-bit-avr-microcontroller-atmega640-1280-1281-2560-2561_datasheet.pdf).  
  
@++;  
Gauthier.

closed 2021-01-25 by fra589