fra589/grbl-Mega-5X #125: Can't get 4th axis to work (slaves fine with jumper)

water-rhatt, 2020-05-25:
Whatever I name it, I get no movement on AXIS_4.  Axis 1, 2, and 3 work fine.  They will map to X,Y, and Z however I want including duplicating an axis.  

Axis 4 will slave just fine with a jumper, just not from within grbl. Given that it moves just fine when jumpered to another axis, I think I can rule out wiring, the motor, or the driver.

I am honestly out of troubleshooting ideas.  Any help is welcome.

The cpu_map.h file comes straight from the maker of the 6 axis shield. It is pretty obvious and logical.  I assume some issue there, but it is beyond me to find.  

Here are the relevant parts of my config.h:

```
/*
  config.h - compile time configuration
  Part of Grbl

  Copyright (c) 2017-2018 Gauthier Briere
  Copyright (c) 2012-2016 Sungeun K. Jeon for Gnea Research LLC
  Copyright (c) 2009-2011 Simen Svale Skogsrud

  Grbl is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  Grbl is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with Grbl.  If not, see <http://www.gnu.org/licenses/>.
*/

// This file contains compile-time configurations for Grbl's internal system.
// This file is used to define the number of axix and their names,
// define homing sequences or specific needs, i.e.
// performance tuning or adjusting to non-typical machines.

// IMPORTANT: Any changes here requires a full re-compiling of the source code to propagate them.

#ifndef config_h
#define config_h
#include "grbl.h" // For Arduino IDE compatibility.

// Define CPU pin map and default settings.
// NOTE: OEMs can avoid the need to maintain/update the defaults.h and cpu_map.h files and use only
// one configuration file by placing their specific defaults and pin map at the bottom of this file.
// If doing so, simply comment out these two defines and see instructions below.
#define DEFAULTS_GENERIC
#define CPU_MAP_2560_INITIAL

// To use with RAMPS 1.4 Board, comment out the above defines and uncomment the next two defines
//#define DEFAULTS_RAMPS_BOARD
//#define CPU_MAP_2560_RAMPS_BOARD

// Serial baud rate
// #define BAUD_RATE 230400
#define BAUD_RATE 115200

// Axis array index values. Must start with 0 and be continuous.
#ifdef DEFAULTS_RAMPS_BOARD
  // 4, 5 & 6 axis support only for RAMPS 1.4 (for the moment :-)...)
  #define N_AXIS 5            // Number of axes
  #define N_AXIS_LINEAR 3     // Number of linears axis
#else
  #define N_AXIS 4 // Number of axes = 3 if not DEFAULTS_RAMPS_BOARD
  #define N_AXIS_LINEAR 4
#endif

#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
[... 167 lines cut ...]
  #define SPINDLE_OCRA_REGISTER   OCR4A // 16-bit Fast PWM mode requires top reset value stored here.
  #define SPINDLE_OCRA_TOP_VALUE  0x0400 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

  // Define spindle output pins.
  #define SPINDLE_PWM_DDR		DDRH
  #define SPINDLE_PWM_PORT  PORTH
  #define SPINDLE_PWM_BIT		4 // MEGA2560 Digital Pin 7

#endif

/* 
#ifdef CPU_MAP_CUSTOM_PROC
  // For a custom pin map or different processor, copy and edit one of the available cpu
  // map files and modify it to your needs. Make sure the defined name is also changed in
  // the config.h file.
#endif
*/

#endif

```

And finally, pinout schematics:

[GRBLDuino Mega Shield pinout.pdf](https://github.com/fra589/grbl-Mega-5X/files/4678743/GRBLDuino.Mega.Shield.pinout.pdf)

fra589, 2020-05-26:
Hi @water-rhatt,  
  
The 5X version of grbl-Mega is based on the RAMPS version. The old section `#define DEFAULTS_GENERIC & #define CPU_MAP_2560_INITIAL` [stay in sources for historical](https://github.com/fra589/grbl-Mega-5X/issues/119) and it can't be used for more than 3 axis.  
  
For 4 axis and more, you need to use the RAMPS section:  
```
#define DEFAULTS_RAMPS_BOARD
#define CPU_MAP_2560_RAMPS_BOARD
```
  
The cpu_map.h is a little bit more complicated and it's describe in the wiki here:  
https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h  

@++;  
Gauthier.

water-rhatt, 2020-05-28:
That worked.  So just to be clear, ANY configuration for more than 3 axis is under the "Ramps" section, regardless of whether you are using a RAMPS shield or something else. The "Generic" isn't to be used EVER, and is just useless vestigial code like nipples on a man.

I'm sure others have made this suggestion, but perhaps that section could be simply renamed to "MULTI_AXIS_BOARD" or something. As it is it appears to apply only to RAMPS 1.4 boards, and that leads to unnecessary confusion.

closed 2020-05-28 by water-rhatt