fra589/grbl-Mega-5X #128: automation

luk1987, 2020-06-18:
Hello
I use his fw and I am very satisfied with it now I need to add some photocells and various controls before launching the program.
Let me explain better I would like to create a real automation
1 piece load
2 presence of work piece (execution grbl5x)
3 piece unloading
I created the sketch for this management but now I should configure it with its fw where I can insert it should you create a library?
Thanks for the attention

fra589, 2020-06-20:
Hello @luk1987,  
I may not have understood your problem exactly, but I think you are on the wrong direction thinking that an event should or could start or stop the execution of Grbl.  
Grbl is not a simple program that you could start or stop like simple software on your computer. On your computer, the operating system (Windows, Linux, Mac OS, ...) which is always running as soon as the computer is turned on, and which is responsible for starting or stopping softwares.  
Grbl is the operating system and the Gcode programs are the software. This means that it is the Gcode programs and not Grbl itself that can be started or stopped.  
Grbl is always in execution as soon as the Arduino board is powered up and can only stop by powering down.  
The operating principle of Grbl is to launch the execution of each line of Gcode as soon as they arrive on the serial interface.  
It is within the main Grbl loop (located between line 74 (`for (;;) {`) and line 166 (`}`) of the protocol.c file that everything happens.  
This loop can only be stopped when Grbl crashes or when Grbl is reset.
It is the role of the interface program (cn5X, GrblGru, CNCjs, Universal Gcode Sender, ...) which is responsible for managing the start or stop of a complete Gcode program by starting or stopping streaming to the serial port.  
The best solution is, in my opinion, not to modify Grbl, but to use the interface software to automate the start of the streaming by adding the hardware and the sensors for the presence detection of the working piece.  
  
@++;  
Gauthier.

luk1987, 2020-06-20:
hi thanks for the quick reply ... my intent is not to stop or start
grbl but to start as the gcode says after reading some sensors ... I
noticed that everything takes place in the protocol.c I wanted to
expand it by importing a library (which reads all the sensors for the
load) which I would then use in the protocol for the start cycle ... I
also wanted to insert a for loop with the numbers of cycles to be done
in order to always repeat the same g code create a software in c sharp
for the management unfortunately I don't know python well to be able
to change its which I consider excellent. do you think it can work?

luk1987, 2020-06-20:
I wanted to add these checks in protocol.c
row 191-196

closed 2021-01-25 by fra589