fra589/grbl-Mega-5X #129: 4 linear axis

qiwalker, 2020-06-26:
hi...
can i make 4 linear, i mean the A axis for linear movement not rotary, if it can could u please tell me how to set it

thankyou verymuch

fra589, 2020-06-27:
Hi @qiwalker,

To get started, it's often enough to search a little and/or look...
[image]
This extract from the config.h file seems rather explicit to me...
So, you can use on line number 54 in config.h:
`#define N_AXIS_LINEAR 4     // Number of linears axis
`
Then, it will be more standard to use a standard linear name for the 4th axis, U, V or W, depending on the direction of your 4th axis rather than A. So you can use (always in config.h): `#define AXIS_4_NAME 'U' // Or V, or W: letter of 4th axis linear` 
cf. http://linuxcnc.org/docs/2.6/html/gcode/machining_center.html#_axes

@++;
Gauthier.

qiwalker, 2020-07-01:
hi @fra589 

thank you very much, machine configuration working, anyway is there any pin for emergency button ?

fra589, 2020-07-01:
Hardware emergency button have to be connected to the reset switch (pin A9).  
See https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout
@++;
Gauthier.

qiwalker, 2020-07-01:
Hi @fra589 

thanks for ur reply,
one more question, i cannot send X0
ie : origin machine position is x0 then i send x100 and then i want to go back to original position when i send x0 machine not moving but when i send x-100 then machine back to original poisition is it normal ?

thanks
rizki

fra589, 2020-07-03:
http://linuxcnc.org/docs/2.7/html/gcode/coordinates.html

CatalinCambosie, 2020-07-19:
Hi,
Please be so kind and tell me   for a Lowrider with  MKS GEN L (Mega 2560) it's ok with the original config.h, or I have to change something for dual axissteppers/drivers for X and Y. If not, how can I change it.  
Many thanks in advance,
Catalin

fra589, 2020-08-31:
Hi @CatalinCambosie,  
  
Please, read the docs :  
https://github.com/gnea/grbl/wiki  
https://github.com/fra589/grbl-Mega-5X/wiki  

Read the config.h and cpu_map.h comments.  

Finally, if you have other question than "do the study for me and give me the result for free", open your own issue.  
  
@++;  
Gauthier.

closed 2020-08-31 by fra589