fra589/grbl-Mega-5X #130: Homing on cloned axis

RaphaelDives, 2020-07-15:
Hey there,
just a very short Question... I'm pretty shure I stumbled across the solution, but I can't find it any more:
I have cloned X and Y axis on my MPCNC and upgraded to new TMC Driver. Now when I start homing the cloned axis stops moving after some cm and only one axis moves on (as long as it is possible... :D). This is the same for X and Y.
The reason is very trivial for shure, but drives me crazy at the Moment.
Thank You very much!

fra589, 2020-07-15:
> The reason is very trivial for shure

Probably trivial... But without any information (grbl-Mega-5X version, config, ...)...  
I am not a soothsayer  

@++;  
Gauthier.

RaphaelDives, 2020-07-16:
Hey, 
sorry, I considered this error so apparent that I didn't think of including further Information... :/
To be honest, I'm running a quite old version, that worked fine with my DRV8825 drivers but not with the TMC2208.
So I think I will test the current version, before going further. 
I will report the results! :)

RaphaelDives, 2020-07-24:
Hello,
sorry for bothering. I installed the current version - the error stays the same. One of the axis will eventually stop working while homing.

But I did a lot of homing for testing and this behaviour seems rather random to me. Sometimes it is the "normal" axis and sometimes the cloned. And eventually it works withaout any error. So I think the problem is more connected to the switches or cables or whatever… After homing everything works fine!

So I will close this topic. 
Thank You very much and keep on working! :)
Cheers
Raphael

closed 2020-07-24 by RaphaelDives