fra589/grbl-Mega-5X #131: QUESTION how to activate deactivate a relay using Z up down

sinfocomp, 2020-07-19:
Is it possible to get a pin high when z moves up and low when it goes down ?
i tryed using the Z DIR pin but seems it does not stay low or high after move completes
Thanks

fra589, 2020-07-19:
Hi @sinfocomp,  
  
It seems to me pretty difficult, since movement are calculated depending on gcode, speed and acceleration, then motors impulsions are stored by the planner. Then, it's a timer interrupt wich send impulsion on all motors pins, without the known of the considered axis.  
  
@++;  
Gauthier.

erosenst, 2020-12-14:
You can add stepper.c uint8_t last_dir that is triggered in main step ISR and keeps some pinpair up/down based st.dit_outbits & st.step_outbits. Last_dir is subtituded only when st.step_outbits==1. Not very hard task to do but may not include official 5X..?
 What is reason you need this functionality..?

fra589, 2021-01-25:
No response from the requester for this solution for several months ... :-(
I guess he fixed his problem and I close this problem.

closed 2021-01-25 by fra589