fra589/grbl-Mega-5X #132: Frequent crashes, limit switches cannot be used

wwwlxs, 2020-07-20:
My board is mega2560+ramps1.4, grbl1.1n, the limit switch is turned on
（ enable the feature in cpu_map.h (line 321) by removing the comment (//) before the sentence:
#define ENABLE_RAMPS_HW_LIMITS
）,
 it often crashes. After restarting, the settings （"$$"） are automatically replaced with incorrect settings. And the settings are different each time.

before:

> 
```

> $$
< $0=6           (Sets time length per step. Minimum 3usec.)
< $1=255         (Sets a short hold delay when stopping to let dynamics settle before disabling steppers. Value 255 keeps motors enabled with no delay.)
< $2=0           (Inverts the step signal. Set axis bit to invert (00000ZYX).)
< $3=8           (Inverts the direction signal. Set axis bit to invert (00000ZYX).)
< $4=0           (Inverts the stepper driver enable pin signal.)
< $5=0           (Inverts the all of the limit input pins.)
< $6=0           (Inverts the probe input pin signal.)
< $10=2          (Alters data included in status reports.)
< $11=0.010      (Sets how fast Grbl travels through consecutive motions. Lower value slows it down.)
< $12=0.002      (Sets the G2 and G3 arc tracing accuracy based on radial error. Beware: A very small value may effect performance.)
< $13=0          (Enables inch units when returning any position and rate value that is not a settings value.)
< $20=0          (Enables soft limits checks within machine travel and sets alarm when exceeded. Requires homing.)
< $21=1          (Enables hard limits. Immediately halts motion and throws an alarm when switch is triggered.)
< $22=1          (Enables homing cycle. Requires limit switches on all axes.)
< $23=31         (Homing searches for a switch in the positive direction. Set axis bit (00000ZYX) to search in negative direction.)
< $24=10.000     (Feed rate to slowly engage limit switch to determine its location accurately.)
< $25=1000.000   (Seek rate to quickly find the limit switch before the slower locating phase.)
< $26=255        (Sets a short delay between phases of homing cycle to let a switch debounce.)
< $27=5.000      (Retract distance after triggering switch to disengage it. Homing will fail if switch isn't cleared.)
< $30=1000       (Maximum spindle speed. Sets PWM to 100% duty cycle.)
< $31=0          (Minimum spindle speed. Sets PWM to 0.4% or lowest duty cycle.)
< $32=0          (Enables laser mode. Consecutive G1/2/3 commands will not halt when spindle speed is changed.)
< $100=200.000   (X-axis travel resolution in steps per millimeter.)
< $101=200.000   (Y-axis travel resolution in steps per millimeter.)
< $102=800.000   (Z-axis travel resolution in steps per millimeter.)
< $103=8.889     (A-axis travel resolution in steps per degre.)
< $104=8.889     (B-axis travel resolution in steps per degre.)
< $110=4000.000  (X-axis maximum rate. Used as G0 rapid rate.)
< $111=4000.000  (Y-axis maximum rate. Used as G0 rapid rate.)
< $112=2000.000  (Z-axis maximum rate. Used as G0 rapid rate.)
< $113=4000.000  (A-axis maximum rate. Used as G0 rapid rate.)
< $114=4000.000  (B-axis maximum rate. Used as G0 rapid rate.)
< $120=200.000   (X-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $121=200.000   (Y-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $122=300.000   (Z-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $123=500.000   (A-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $124=500.000   (B-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $130=600.000   (Maximum X-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $131=600.000   (Maximum Y-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $132=600.000   (Maximum Z-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)

```

 After restarting：
```
> * RESET
< Grbl 1.1n ['$' for help]
* Version 1.x
* Read grbl settings, hide response from '$$', '$#'
* Buffer size was manually set to 127 bytes!
* Check [Setup - Flow control]
> $$
< $0=0           (Sets time length per step. Minimum 3usec.)
< $1=255         (Sets a short hold delay when stopping to let dynamics settle before disabling steppers. Value 255 keeps motors enabled with no delay.)
< $2=24          (Inverts the step signal. Set axis bit to invert (00000ZYX).)
< $3=112         (Inverts the direction signal. Set axis bit to invert (00000ZYX).)
< $4=0           (Inverts the stepper driver enable pin signal.)
< $5=0           (Inverts the all of the limit input pins.)
< $6=0           (Inverts the probe input pin signal.)
< $10=0          (Alters data included in status reports.)
< $00=200.000    ( no information found '00')
< $101=200.000   (Y-axis travel resolution in steps per millimeter.)
< $102=800.021=1 (Z-axis travel resolution in steps per millimeter.)
< $22=1          (Enables homing cycle. Requires limit switches on all axes.)
< $23=4          (Homing searches for a switch in the positive direction. Set axis bit (00000ZYX) to search in negative direction.)
< $24=10.0$110=2000.000 (Feed rate to slowly engage limit switch to determine its location accurately.)
< $111=2000.
$27=10.000 (Y-axis maximum rate. Used as G0 rapid rate.)
< $30=1000       (Maximum spindle speed. Sets PWM to 100% duty cycle.)
< $31=0          (Minimum spindle speed. Sets PWM to 0.4% or lowest duty cycle.)
< $32=1          (Enables laser mode. Consecutive G1/2/3 commands will not halt when spindle speed is changed.)
< $100=200.000   (X-axis travel resolution in steps per millimeter.)
< $101=200.000   (Y-axis travel resolution in steps per millimeter.)
< $102=800.000   (Z-axis travel resolution in steps per millimeter.)
< $103=8.889     (A-axis travel resolution in steps per degre.)
< $104=8.889     (B-axis travel resolution in steps per degre.)
< $110=2000.000  (X-axis maximum rate. Used as G0 rapid rate.)
< $111=2000.000  (Y-axis maximum rate. Used as G0 rapid rate.)
< $112=2000.000  (Z-axis maximum rate. Used as G0 rapid rate.)
< $113=8000.000  (A-axis maximum rate. Used as G0 rapid rate.)
< $114=8000.000  (B-axis maximum rate. Used as G0 rapid rate.)
< $120=200.000   (X-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $121=200.000   (Y-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $122=200.000   (Z-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $123=50.000    (A-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $124=50.000    (B-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $130=500.000   (Maximum X-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $131=500.000   (Maximum Y-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $132=500.000   (Maximum Z-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $133=1000.000  (Maximum A-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $134=1000.000  (Maximum B-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< ok
```

fra589, 2020-07-20:
Hi @wwwlxs,  
  
Strange behavior... 
It looks like a problem that happens quite commonly when the number of axes or when different versions or parameters are successively flashed...  
In general, this is corrected with the Grbl command: `$RST=*` which reformats the entire parameter memory with the default values.  
After this, you can put and store your own parameters, it should be stable.  
  
@++;  
Gauthier.

wwwlxs, 2020-07-20:
> Hi @wwwlxs,
>
> Strange behavior...
> It looks like a problem that happens quite commonly when the number of axes or when different versions or parameters are successively flashed...
> In general, this is [...]

Thank you for your reply, I set it up, but there are new problems，Sometimes there is a problem sometimes no problem:

```
<error:15 Jog target exceeds machine travel. Jog command has been ignored.
> '0x85' Jog Cancel
```
new setting：
``
```
> $$
< $0=10          (Sets time length per step. Minimum 3usec.)
< $1=254         (Sets a short hold delay when stopping to let dynamics settle before disabling steppers. Value 255 keeps motors enabled with no delay.)
< $2=0           (Inverts the step signal. Set axis bit to invert (00000ZYX).)
< $3=8           (Inverts the direction signal. Set axis bit to invert (00000ZYX).)
< $4=0           (Inverts the stepper driver enable pin signal.)
< $5=0           (Inverts the all of the limit input pins.)
< $6=0           (Inverts the probe input pin signal.)
< $10=0          (Alters data included in status reports.)
< $11=0.020      (Sets how fast Grbl travels through consecutive motions. Lower value slows it down.)
< $12=0.002      (Sets the G2 and G3 arc tracing accuracy based on radial error. Beware: A very small value may effect performance.)
< $13=0          (Enables inch units when returning any position and rate value that is not a settings value.)
< $20=1          (Enables soft limits checks within machine travel and sets alarm when exceeded. Requires homing.)
< $21=1          (Enables hard limits. Immediately halts motion and throws an alarm when switch is triggered.)
< $22=1          (Enables homing cycle. Requires limit switches on all axes.)
< $23=31         (Homing searches for a switch in the positive direction. Set axis bit (00000ZYX) to search in negative direction.)
< $24=100.000    (Feed rate to slowly engage limit switch to determine its location accurately.)
< $25=500.000    (Seek rate to quickly find the limit switch before the slower locating phase.)
< $26=250        (Sets a short delay between phases of homing cycle to let a switch debounce.)
< $27=5.000      (Retract distance after triggering switch to disengage it. Homing will fail if switch isn't cleared.)
< $30=12000      (Maximum spindle speed. Sets PWM to 100% duty cycle.)
< $31=550        (Minimum spindle speed. Sets PWM to 0.4% or lowest duty cycle.)
< $32=0          (Enables laser mode. Consecutive G1/2/3 commands will not halt when spindle speed is changed.)
< $100=200.000   (X-axis travel resolution in steps per millimeter.)
< $101=200.000   (Y-axis travel resolution in steps per millimeter.)
< $102=800.000   (Z-axis travel resolution in steps per millimeter.)
< $103=8.889     (A-axis travel resolution in steps per degre.)
< $104=8.889     (B-axis travel resolution in steps per degre.)
< $110=750.000   (X-axis maximum rate. Used as G0 rapid rate.)
< $111=750.000   (Y-axis maximum rate. Used as G0 rapid rate.)
< $112=750.000   (Z-axis maximum rate. Used as G0 rapid rate.)
< $113=1440.000  (A-axis maximum rate. Used as G0 rapid rate.)
< $114=1440.000  (B-axis maximum rate. Used as G0 rapid rate.)
< $120=50.000    (X-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $121=50.000    (Y-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $122=50.000    (Z-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $123=50.000    (A-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $124=50.000    (B-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $130=800.000   (Maximum X-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $131=800.000   (Maximum Y-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $132=50.000    (Maximum Z-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $133=360.000   (Maximum A-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $134=180.000   (Maximum B-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< ok

```
After I reset and start it, `$10=2,` I start it again` $10=2` automatically becomes` $10=0`, but I did not change $10，That is, the settings will still be changed automatically and irregularly

wwwlxs, 2020-07-20:
The homing cycle is still unavailable, and there is no rebound to slowly engage limit switch after quickly find touching the limit switch for the first time, but an alarm
```
< ALARM:9 	 or check wiring.
< ok
* RESET

```

glockpod, 2020-09-05:
I have been seeing a similar problem, but in my case, it is caused by having a maximum velocity, and a maximum step/mm that result in more than 30K steps per second. 

Using your settings above for the Z axis:

800steps/mm x 2000mm/min x 1min/60sec => 26666.67steps/second

Which is right near the 30Ksteps/second limit... (Which I think is more an adhoc limit than a scientific limit.)

Every time I move with such a high step speed, the instant the motor accelerates above the limit, GRBL crashes hard, and rewrites the entire EEPROM space with garbage.  Even $RST=* was unable to repair the damage to one of my Arduino boards... which remains unusable to this day.

It would appear that it might be a good time for the author to decide if the internal limiter on maximum step rate is safe to uncomment and use.  I don't believe it is.

In the mean time, you might want to try cutting your maximum mm/minute rate in half and see if your problem goes away.

-Chuck

RaphaelDives, 2020-09-07:
The same here!
had the problem when I switched my drivers to TMC2209 - I thougt the more steps the better. ;)
In my case, moving one axis per time worked, two axis at the same time resulted in a similar behaviour. I also had the strange values in the configuration....

I had no idea about maximum steps, but I set the jumper on my ramps board to less microstepping and the errors were gone...

Good luck!
Raphael

glockpod, 2020-09-11:
Hi Raphael,

I got into my problem exactly the same way.  I was using the old red motor drivers, and needed to increase my resolution, as I was using a motor direct drive, and 16x microstepping wasn't quite fine enough.... So, I bought some of the purple TMC2209 based motor driver boards to try.  I doubled the mm/minute number, and doubled the mm maximum travel settings, and the first time I moved the motor: <<BOOM!!>>  my machine made a loud crashing sound, and was left squealing and quivering.

The problem is the MEGA 256 processor isn't fast enough to get the interrupt service routines that drive the motors done before the next motion step happens, and the stack grows past its limits, eating the flash ram.  I believe the only to restore function to the Arduino is to reflash its boot loader, and start it up like the factory does.

Anyway, it is abundantly clear that the author is gone, and this project has been abandoned.  Which is a shame.

-Chuck

fra589, 2020-09-11:
> Anyway, it is abundantly clear that the author is gone, and this project has been abandoned. Which is a shame.

Hi @glockpod,

The project is not abandoned, but I don't always have an answer to all the questions asked.
I don't always respond, especially when other people answer with good leads. :-)

Yes, the Hardware is limited, so, in many case with this hardware, it's necessary to choose between speed or precision and make a compromise.

@++;
Gauthier.

glockpod, 2020-09-11:
A slight mistake, not the mm max travel setting, but rather the steps/mm.

The settings for the new purple stepper motor driver board resulted in about 38KHz step rate, which trashes the board.

glockpod, 2020-09-11:
I am glad that you are still here.

Yes, the hardware is limited, but then so is my requirement... a coil winding machine.

The problem is, we don't know what the maximum rate is, and it seems to change with programming options, the number of motors running simultaneously, and who knows what else?

Trying to figure out the maximum rate trashes the board most thoroughly when you exceed the maximum.  As I said earlier, I have one board that won't come back even after a fresh software load, and using the RST=* command.  It just squeals at me.

There is a compile time option to cause the board to verify that the stepping rate isn't exceeded.  Can it be made to work?

-Chuck

fra589, 2020-09-14:
Hi @glockpod,  
  
Compile time option to verify the stepping rate isn't exceeded is practically impossible to code.
The max step frequency can significantly vary with as many parameters that is impossible for me to know... In addition, the final max frequency will depend about speed, steps/mm, etc. parameters defined at run time. So, a compile time warning or error isn't possible to define.

Have a look about those discuss on this subject :
https://github.com/grbl/grbl/issues/41 (pretty old, but intressting)  
https://github.com/fra589/grbl-Mega-5X/issues/56  

See also in the stepper.c file, the max step frequency ans speed limitation are explained in comments...  
  
@++;  
Gauthier.

glockpod, 2020-09-15:
Hi Gauthier,

Although I wrote my first interrupt driven stepper motor driver some 35 years ago, I don't know much about how GRBL works.

My suspicion is that GRBL is suffering from a stack blow up caused by a motor handling ISR not being able to complete its task before the next interrupt.  There are lots of ways of preventing this from becoming a problem, but each has its drawbacks.  One is to leave the motor interrupt masked until the motor ISR has completed.  Another is to count the entries to the motor interrupt, and the exits from the motor interrupt, and if there are more entries, than exits, shut down the motor interrupts, stopping the motors.

As I said, I know virtually nothing about how GRBL works, and really haven't the time to dig into it right now.

-Chuck

closed 2020-10-02 by fra589