fra589/grbl-Mega-5X #134: More general questions to add support to our Commander Software

Slixxor, 2020-07-23:
Does the 4th axis support homing?

How can homing direction be set and equivalent table?

Is the 4th axis (assuming rotary) in degrees or mm/inches?

fra589, 2020-07-23:
> Does the 4th axis support homing?

Yes it support it, if thi is correctly configured at compile time (HOMING_CYCLE_* in config.h.

> How can homing direction be set and equivalent table?  

https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration
See $23

> Is the 4th axis (assuming rotary) in degrees or mm/inches?

Depending of the N_AXIS_LINEAR definition in config.h, the 4th axis will be in angle unit or in length unit...
For example, if N_AXIS_LINEAR = 3, all the axis >3 will be rotational axis, so the 4th is rotational. If N_AXIS_LINEAR = 4, the 4th axis will be linear and his unit will be mm/inch.

@++;
Gauthier.

Slixxor, 2020-07-26:
perfect thanks.

closed 2020-07-26 by Slixxor