fra589/grbl-Mega-5X #137: Inverted (NC) limit switches ($5=1) not working; simple fix (from April code change)

cmidgley, 2020-08-15:
Hi - in the April code update to limits.c, it appears a small bug got in there and stops inverted limit switches from working (they always report as triggered, independent of the pin value).  The problem can be found at line 221-225 of limits.c, which has:

```
    if (bit_istrue(settings.flags,BITFLAG_INVERT_LIMIT_PINS)) {
      return(~(limit_state_max & limit_state_min));
    } else {
      return(limit_state_max | limit_state_min);
    }
```

Notice in the first return (line 222) it is using "&" whereas in 224 it uses "|".  OR makes sense (merge the bits), whereas AND does not (and results in 0, and after inversion, all bits on).  The fix is simply to change line 222 to:

```
      return(~(limit_state_max | limit_state_min));
```

I've briefly tested this and it appears to be working with NC switches.  Sorry I've not submitted a pull request - I'm not currently set up a fork for this and thought this fix was small enough that this would be sufficient.

fra589, 2020-08-27:
Hi @cmidgley,  
  
Sorry for my response delay, I was on vacation... :smiley:  
  
No, the & binary operator at line 222 is the correct one (don't forget the negation "~" operator!).  
Given that if you configure $5=1, BOTH min and max limit switches must be closed (Normaly Closed switchs) so that the limit switch is not triggered.
If you have only one limit switch (max switch for example), you must short circuit the other pin (the pin of the min switch in the example) to the ground to simulate the missing normaly closed switch.  
  
@++;  
Gauthier.

closed 2020-08-31 by fra589

cmidgley, 2020-08-31:
Hi Gauthier - yes, you are correct, it was my mistake.  I have undone that change, added the wires to close the max limit switches and everything is working great.  Sorry to have bothered you on this one, I should have examined that code more closely.  Appreciate all your time and effort on this project.