fra589/grbl-Mega-5X #138: Relocating SPINDLE pin to D11

Fusseldieb, 2020-08-18:
I've been trying to change the D8 pin to D11 to be able to drive my laser with my new SKR board, but I'm a little bit stuck.

I've changed the lines to following, researching (a lot) on the internet, but I think I still got PORTs, registers and such things all wrong. It still works, but the question is... Does it well? And how should I fix this?

```
  // Advanced Configuration Below You should not need to touch these variables
  // Set Timer up to use TIMER4B which is attached to Digital Pin 8 - Ramps 1.4 12v output with heat sink
  #define SPINDLE_PWM_MAX_VALUE     1024.0 // Translates to about 1.9 kHz PWM frequency at 1/8 prescaler
  #ifndef SPINDLE_PWM_MIN_VALUE
    #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
  #endif
  #define SPINDLE_PWM_OFF_VALUE     0
  #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)

  //Control Digital Pin 6 which is Servo 2 signal pin on Ramps 1.4 board
  #define SPINDLE_TCCRA_REGISTER    TCCR1A
  #define SPINDLE_TCCRB_REGISTER    TCCR1B
  #define SPINDLE_OCR_REGISTER      OCR1A
  #define SPINDLE_COMB_BIT          COM1A1

  // 1/8 Prescaler, 16-bit Fast PWM mode
  #define SPINDLE_TCCRA_INIT_MASK ((1<<WGM40) | (1<<WGM41))
  #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))
  #define SPINDLE_OCRA_REGISTER   OCR1B // 16-bit Fast PWM mode requires top reset value stored here.
  #define SPINDLE_OCRA_TOP_VALUE  0x0400 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

  // Define spindle output pins.
  #define SPINDLE_PWM_DDR   DDRB
  #define SPINDLE_PWM_PORT  PORTB
  #define SPINDLE_PWM_BIT   5 // MEGA2560 PWM Pin 11
```

If someone could shed me a light and tell me which are the right ports and registers for D11, it would be immensely helpful!

Thanks

fra589, 2020-08-20:
Hi @Fusseldieb,  
  
See : https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h  
  
@++;  
Gauthier.  

They have eyes, and do not know how to look...

Fusseldieb, 2020-08-25:
> Hi @Fusseldieb,
> 
> See : https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h
> 
> @++;
> Gauthier.
> 
> They have eyes, and do not know how to look...

Thanks for the reply! I was thinking about those TCCRA, TCCRB, OCR, OCRA registers. I didn't know how to set them. I couldn't find much on the matter.

In the end, I just left the default port. But it would be awesome having reference for future searchers :)

ravx4, 2020-11-12:
Hello everyone,
I've met similar issue. I'm trying to use MKS 1.4 board for a laser project and the only convenient pin avaliable for laser PWM is pin D6, so I'd like to remap pin D8 (RAMPS) to  pin D6 (MKS 1.4).
 
(change of:
 // Define spindle output pins.
  #define SPINDLE_PWM_DDR   DDRH
  #define SPINDLE_PWM_PORT  PORTH
  #define SPINDLE_PWM_BIT   5 // MEGA2560 Digital Pin 8

to

  // Define spindle output pins.
  #define SPINDLE_PWM_DDR   DDRH
  #define SPINDLE_PWM_PORT  PORTH
  #define SPINDLE_PWM_BIT   3 // MEGA2560 Digital Pin 6

as @fra589 mentioned via linked wiki is not worky worky)

EDIT:
After some compare and trial/error I've managed to blink led at pin D6 with M3/M4 command but led is switching either 100% or 0% despite trying M3/M4 at minimum/medium/maximum values.
here's the code i've came up with:

  //Control Digital Pin 6 which is Servo 2 signal pin on Ramps 1.4 board
  #define SPINDLE_TCCRA_REGISTER    TCCR4A
  #define SPINDLE_TCCRB_REGISTER    TCCR4B
  #define SPINDLE_OCR_REGISTER      OCR4A (changed from OCR4C)
  #define SPINDLE_COMB_BIT          COM4A1 (changed from COM4C1)

  // 1/8 Prescaler, 16-bit Fast PWM mode
  #define SPINDLE_TCCRA_INIT_MASK ((1<<WGM40) | (1<<WGM41))
  #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))
  #define SPINDLE_OCRA_REGISTER   OCR4A // 16-bit Fast PWM mode requires top reset value stored here. (tried OC4B, no difference in led behaviour)
  #define SPINDLE_OCRA_TOP_VALUE  0x0400 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

  // Define spindle output pins.
  #define SPINDLE_PWM_DDR   DDRH
  #define SPINDLE_PWM_PORT  PORTH
  #define SPINDLE_PWM_BIT   3 // MKS Digital Pin 6 (change #define SPINDLE_PWM_BIT   5 to #define SPINDLE_PWM_BIT   3)

Please, can anyone give me some clue how to change the pin mapping and register values so I can get pin D6 work with adjustable PWM?  Right now im digging through atmega 2560 datasheet trying to understand anything about registers and timers but ended up more confused and frustrated than before :(

fra589, 2020-11-12:
Hi @ravx4,  
  
In order to the spindle output pin definition, you also need to change the register SPINDLE_OCR_REGISTER definition for the PWM working:

Arduino Pin | Register
-- | --
2 | OCR3B
3 | OCR3C
4 | OCR4C
5 | OCR3A
6 | OCR4A
7 | OCR4B
8 | OCR4C
9 | OCR2B
10 | OCR2A
11 | OCR1A
12 | OCR1B
13 | OCR0A
44 | OCR5C
45 | OCR5B
46 | OCR5A
  
@++;  
Gauthier

ravx4, 2020-11-12:
Thank You very much for reply,
as mentioned above, I've made changes as below:
#define SPINDLE_OCR_REGISTER OCR4A (changed from original OCR4C)
#define SPINDLE_COMB_BIT COM4A1 (changed from original COM4C1)
#define SPINDLE_PWM_BIT 3 // MKS Digital Pin 6 (change #define SPINDLE_PWM_BIT 5 to #define SPINDLE_PWM_BIT 3)

But right now with these settings, there's no difference in led brightness when I send M3 S1 or M3 S100 

($30=100 (Maximum spindle speed. Sets PWM to 100% duty cycle.)
$31=0(Minimum spindle speed. Sets PWM to 0.4% or lowest duty cycle.)

should I also change register in this line? 
#define SPINDLE_OCRA_REGISTER OCR4A // 16-bit Fast PWM mode requires top reset value stored here.
original register is OCR4A which already correspond to pin D6, so i'm confused.

ravx4, 2020-11-12:
Hello @fra589 , I've loaded your code, now D6 is indicating pwm changes as supposed, so im very gratefull for Your help, many thanks :D
If you dont mind i have two more questions:
Is fact thad D6 is 8bit is going to have any impact on performance? for example speed or something else?
How can I determine **ocra_register value**, for example if i want to try pwm on some other pins?

fra589, 2020-11-13:
HI @ravx4,  
  
The 8 bits PWM have no impact on the speed performance compared to the 16 bits one. The only drawback of 8 bits PWM in our case is that there is only 255 intermediate value compared to the 1024 one with 16 bits wich is then more precise.  
  
The bible that holds all the truth about the ATmega2560 is the Microship datasheet: http://ww1.microchip.com/downloads/en/DeviceDoc/ATmega640-1280-1281-2560-2561-Datasheet-DS40002211A.pdf  
  
But [Googleing for "pwm atmega2560"](https://www.google.com/search?client=firefox-b-d&q=pwm+atmega2560) can rapidly give usefull information as we are not the first to ask this question 😄   
  
@++;  
Gauthier.

ravx4, 2020-11-13:
Hi @fra589 
http://ww1.microchip.com/downloads/en/DeviceDoc/ATmega640-1280-1281-2560-2561-Datasheet-DS40002211A.pdf thats exactly the document im digging through :P anyway maybe someday i'll understand this silicone elvish language ;P thanks again for help ;)

closed 2020-11-15 by fra589

dodidada, 2021-10-20:
> Hi @dodidada<https://github.com/dodidada>,
>
> The Grbl firmware software is not a standard Arduino IDE sketch.
> The Arduino sketch is only an empty sketch for loading and upload Grbl to the card (procedure here : https://github.com/fra589/grbl-Mega-5X/wiki/Compiling-grbl-Mega-5X)
>
> You need edit the config.h (and another Grbl files) in their place in the library folder.
> You can find or change the location of your sketchbook folder at File > Preferences > Sketchbook location.
> Be careful to have only one version of Grbl under the library directory.
> I advise you to use another editor than the Arduino IDE to modify the Grbl code, this allows you to keep the "GrblUpload" sketch loader in the IDE and use it for each update you try.
>
> @++;
> Gauthier.

thank you so much for fast reply I really appreciate your effort and great job you are doing for the cumunitiy, have a successful day bro.

dodidada, 2021-10-20:
and yes forgot to mention that I feed the spindle with external 12v 30 amp psu