fra589/grbl-Mega-5X #139: 5-axis post processor for fusion 360

abdulrahilsheikh, 2020-08-27:
I've tried editing existing grbl post-processor in fusion 360 but it doesn't turn out to be good.
My g-code just got messed up.
Do we have any 5-axis post processor for fusion 360 which works well for grbl-5axis
Thanks in advance.
All the best for future

fra589, 2020-08-27:
Hi @abdulrahilsheikh,  
As I sayed (in french to) in this issue : https://github.com/fra589/grbl-Mega-5X/issues/107 :
The defense of freedom and the sharing of knowledge being my strongest motivation in writing and sharing my computer skills, I cannot recommend enough that you use open-source software under a free license for your CAD.
The excellent FreeCAD works well with Grbl. I know something about it, I'm the author of the FreeCAD-Grbl post-processor :smiley:   
As for the Fusion360-Grbl post-processors, I wouldn't be of much use, I don't use Fusion and I certainly don't want to use it! The list of reasons is too long to explain to be mentioned here.
Finally, a simple search for "Fusion 360 post-processor" on G**gle led me (among many other results) here: http://wiip.fr/node/8549
It's probably a good place to start doing what you need to do.  
  
@++;  
Gauthier.

closed 2020-08-27 by fra589

AgriMechatronic, 2023-03-12:
Hello, does FreeCAD have support and post processor for 5 axis grbl cnc (mega5x)?
And can we use generated codes from FreeCAD to send to cnc with GrblGru or we need anything else?

Thanks in advance.

fra589, 2023-03-13:
Hi @AgriMechatronic,  
  
FreeCAD does not yet support 5-axis milling, this is in development and progressing slowly.
See this discussion on the FreeCAD forum:
https://forum.freecad.org/viewtopic.php?t=50276  

The FreeCAD -> Grbl post processor is already able to process the 5 axes.  
  
I don't know enough about GrblGru to answer the question about it, but I also wrote cn5X (https://github.com/fra589/cn5X) for driving the Grbl machine and sending GCode. It aims to natively support all grbl-Mega-5X features (up to 6 axes + all additional features).

@++;
Gauthier.