fra589/grbl-Mega-5X #143: Spindle pwm issue

hzxiaoyu, 2020-09-23:
Hi, @fra589 
I am trying to control SPINDLE by GRBL, which before is by manuel. But my servo motor is control by 0-5V not pwm, I can buy some mod can change PWM signal to analogy signal. 
But I want to know if there are have any simpler way?

hzxiaoyu, 2020-09-24:
I tried laser mode, I measured pin 4 ,after m3 is 4.8V, But I changed M3 s100 or M3 s10  with no different.

fra589, 2020-09-24:
Hi @hzxiaoyu,  
  
Pulse Width Modulation, or PWM, is a technique for getting analog results with digital means.  
  
Have à look here:  
https://github.com/fra589/grbl-Mega-5X/blob/edge/doc/script/fit_nonlinear_spindle.py  
This script permit to calculate all PWM parameters, and it's comments explain how PWM spindle work in Grbl and how it can be adjusted.  
  
You can also search for the word PWM in the precedents issues:  
https://github.com/fra589/grbl-Mega-5X/issues?q=pwm+in%3Atitle  
  
@++;  
Gauthier.

hzxiaoyu, 2020-09-24:
@fra589 thank you.
I will try what to do with it.
And another issue, after I updated your last verion, the motor will auto slp when the system is free, I can move axis by hand because my step motor have hand wheel. Only when  I send move x or y or z is normal. I don't know if it is right? where I can config it?

fra589, 2020-09-24:
Please, read the doc https://github.com/fra589/grbl-Mega-5X/wiki ...  
Particulary in your case : https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration#1---step-idle-delay-milliseconds

hzxiaoyu, 2020-09-24:
@fra589 thank you again, I found my setting is $1=225, input wrong by mistake.

closed 2020-10-02 by fra589