fra589/grbl-Mega-5X #146: Problem with <grbl.h>

BosniaCNC, 2020-10-01:
I don't understand what the problem is? Can anyone help?
Thanks
[image]
[image]

RaphaelDives, 2020-10-01:
Looks, like the "library" is not at the correct path?
Hint: the grbl-mega-5x source code goes in the Arduino library folder named "grbl".
Good luck!
Raphael

BosniaCNC, 2020-10-01:
All right! Thanks!

fra589, 2020-10-01:
https://github.com/gnea/grbl/wiki/Compiling-Grbl

closed 2020-10-02 by fra589

BosniaCNC, 2020-10-02:
#include <grbl.h>   =   #include "grbl.h"

fra589, 2020-10-02:
If you have read the compiling doc (link I wrote few hours ago), you have put the grbl source in the standard Arduino library directory. So, the directive `#include <grbl.h>` in the file grblUpload.ino will work fine.

See the doc of the Include Syntax here: https://gcc.gnu.org/onlinedocs/cpp/Include-Syntax.html  
  
@++;  
Gauthier.