fra589/grbl-Mega-5X #147: Homing issue: Only Z axis is homing on my Ramps board 1.4, latest released hex file

TheophileLebrun, 2020-10-02:
Hello,

I flashed the latest released hex file into a mega/ramps1.4, connected stepper drivers for X,Y,Z,A axis and switches for axis X,Y,Z.

I can jog all axis without any issues, I oriented all axis in the positive direction facing the switches, but when i trigger the homing sequence, only the Z axis performs it. It then does nothing until the homing sequence times out.

Very disturbingly, on a couple occasions while trying to perform the homing sequence, i saw the X axis going toward the negative direction and even saw the A axis rotating. I cannot reproduce it systematically,

I replaced the arduino mega, but the behavior stays the same

the $? shows that my switches are well connected and only showup in the report when pressed.

Would anyone know from where the issue could be coming from?

Best regards
Theo

Grbl 1.1l ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 0    (Status report options, mask)
$11 = 0.020    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 500.000    (Homing locate feed rate, mm/min)
$25 = 2000.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 40.000    (X-axis travel resolution, step/mm)
$101 = 34.000    (Y-axis travel resolution, step/mm)
$102 = 300.000    (Z-axis travel resolution, step/mm)
$103 = 8.889   
$104 = 8.889   
$110 = 9000.000    (X-axis maximum rate, mm/min)
$111 = 9000.000    (Y-axis maximum rate, mm/min)
$112 = 300.000    (Z-axis maximum rate, mm/min)
$113 = 1440.000   
$114 = 1440.000   
$120 = 300.000    (X-axis acceleration, mm/sec^2)
$121 = 300.000    (Y-axis acceleration, mm/sec^2)
$122 = 100.000    (Z-axis acceleration, mm/sec^2)
$123 = 100.000   
$124 = 100.000   
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
$133 = 360.000   
$134 = 180.000

fra589, 2020-10-02:
Hi @TheophileLebrun,  
  
Can you post your axis definition (lines 50 to 84 of config.h) and the homing configuration (lines 171 to 199 of the same file config.h) ?  
  
@++;  
Gauthier.

TheophileLebrun, 2020-10-02:
Hello Gauthier,

Thanks for your reply, i did not compile the hex from source, i took the
hex file from the github repo latest release.

I can try to compile from source to see if it changes the situation.

Cordialement,
Theo

fra589, 2020-10-02:
Hello Theo,  
  
I think I understand your problem !  
The latest release available on Github was not the really latest one. In fact, I stopped to give generic HEX file because too many different configurations would be necessary and because there is too many problems like yours with it.  
So, I have just deleted the HEX files still present on Github repo. I'm sorry, I should have done it a long time ago.  
The latest up to date version is only available from the [main grbl-Mega-5X github page](https://github.com/fra589/grbl-Mega-5X).
The Wiki documentation to compile it yourself is here : https://github.com/gnea/grbl/wiki/Compiling-Grbl  

For your need, you have to define in config.h:  
line 53 & 54:  
```
  #define N_AXIS 4            // Number of axes
  #define N_AXIS_LINEAR 3     // Number of linears axis
```
line 172 to 175:  
```
  #if N_AXIS == 4 // 4 axis : homing
    #define HOMING_CYCLE_0 (1<<AXIS_3) // Home Z axis first to clear workspace.
    #define HOMING_CYCLE_1 (1<<AXIS_1) // Home X axis
    #define HOMING_CYCLE_2 (1<<AXIS_2) // Home Y axis
```
=> Suppress the line for homing the 4th axis since you don't have limit switch on it. If you leave it, grbl will timeout trying to find your 4th axis limit switch!  
  
@++;  
Gauthier.

TheophileLebrun, 2020-10-03:
Hello Gauthier,

I followed your steps and it is now working like a charm :)

Thanks a lot!
Cordialement,
Theo

closed 2020-10-03 by TheophileLebrun