fra589/grbl-Mega-5X #149: 4 axis Hotwire CNC on

EhsanBaninajar, 2020-10-28:
Hi,

I'm trying to use Mega5x with a Rmaps1.4 for a 4 axis hotwire CNC (Something similar to https://github.com/fra589/grbl-Mega-5X/issues/121). My problem is the Y axis on the ramps is not working due to faulty circuit. Therefore I need to use X, Z, E0 and E1 for XYUV axes. Is there a way to bypass the Y axis in the codes and use Z axis instead for XY and E0 and E1 for UV? For example can I change the Axis indexing value in order to bypass Y or should I change pins in the CPU_map? 

Thank you,
Ehsan

fra589, 2020-10-29:
Hi @EhsanBaninajar,  
  
You can work around your Y axis problem with a 5 axis axis configuration in config.h, then rename your axis like this :  
```
#define N_AXIS 5            // Number of axes
#define N_AXIS_LINEAR 4     // Number of linears axis
(...)
#define AXIS_1_NAME 'X'
(...)
#define AXIS_2_NAME 'A'
(...)
#define AXIS_3_NAME 'Y'
(...)
#define AXIS_4_NAME 'U'
(...)
#define AXIS_5_NAME 'V'
```
And sure, never use the faulty "A" axis 😃   
Like this, you will be able to avoid modifying the cpu_map.h.  
  
@++;
Gauthier.

fra589, 2021-01-25:
No response from the requester for this solution for almost 3 months ... :-(
I assume he solved his problem and I close this issue.

closed 2021-01-25 by fra589