fra589/grbl-Mega-5X #151: Calibrating step/deg

AngusLawlessDunn, 2020-11-06:
Hi, 

I have one of my axes (A) as a purely rotational axis which takes an input value of degrees (or radians). I am trying to calibrate parameter $103 for this axis and on the configuration wiki page it has listed the parameter can either be steps/mm or steps/deg. My stepper motor is a standard 1.8°/step at 1/18th microstepping meaning 1600 steps per revolution. This means the steps/deg would be 1600/360 = 4.44, however, when I use that value for the parameter $103 and pass the movement A360, I'm not seeing a full rotation. Is there something I'm missing? Any help would really be appreciated!

Thank you,
Angus

fra589, 2020-11-06:
Hi Angus,  
  
1.8° is standard 200 step for 360°. 200x18=3600 (1/18th microstepping), not 1600 steps per revolution...  
But 1/18th don't seems to me a standard value for microstepping?  
  
Don't hesitate to post your full Grbl config ($I & $$ outputs) for this kind of question.  
  
@++;  
Gauthier.

AngusLawlessDunn, 2020-11-06:
Whoops, my apologies I meant 1/8th microstepping! 
200x8=1600 steps per revolution.
I am hoping to be able to send grbl a value in degrees and have it rotate the stepper motor by that amount. Is that feasible?

Thank you,
Angus

fra589, 2020-11-06:
Yes, it's doable. I do it regularly.  
Post the values of your configuration ($ I & $$ outputs) that we try to understand what is wrong for you.  
  
@++;  
Gauthier.

AngusLawlessDunn, 2020-11-26:
Hi Gauthier,
Apologies in the delay in getting back to you. I'm having more trouble now as the A axis won't rotate at all. I can confirm this is not a wiring issue as I am seeing a full rotation of the motor using a simple Arduino sketch:
```
const int stepPin = 26; 
const int dirPin = 28; 
 
void setup() {
  pinMode(stepPin,OUTPUT); 
  pinMode(dirPin,OUTPUT);
}
void loop() {
  digitalWrite(dirPin,HIGH);
  
  for(int x = 0; x < 1600; x++) {
    digitalWrite(stepPin,HIGH); 
    delayMicroseconds(1000); 
    digitalWrite(stepPin,LOW); 
    delayMicroseconds(1000); 
  }
  delay(1000); // One second delay
}
```

Once I upload Grbl-5X however, A can see movement in X, Y, Z axis normally but when passing the command A180 I don't see any response from the motor.

Here is the output from the $ I & $$ commands respectively:
```
[VER:1.1n.20200424:]
[AXS:4:XYZA]
[OPT:VNMCZH,35,255,80]
ok
```
```
$0=10
$1=254
$2=0
$3=0
$4=0
$5=1
$6=0
$10=1
$11=0.020
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=3
$24=100.000
$25=2000.000
$26=0
$27=5.000
$30=12000
$31=550
$32=0
$100=16.667
$101=16.667
$102=200.000
$103=4.444
$110=18000.000
$111=36000.000
$112=5000.000
$113=5000.000
$120=500.000
$121=500.000
$122=100.000
$123=100.000
$130=2000.000
$131=2000.000
$132=2000.000
$133=20000.000
ok
```

I am using 26 for the step pin and 28 for the direction pin as defined by lines 169 and 201 in the 'cpu_map.h'.
Do you have any ideas or suggestions?

Kind regards,
Angus

AngusLawlessDunn, 2020-11-27:
Hi Gauthier,

Apologies for the delay in getting back to you, I had some other issues which I have now solved. It looks like I had just set the value for $103 incorrectly and everything is working now. Passing the command A360 gives one full revolution as expected. Thank you for your help and quick replies!

Angus

closed 2020-11-27 by AngusLawlessDunn

fra589, 2020-11-27:
Hi Angus,  
  
You're welcome 😃   
  
Like what the fact of sending parameters to the issue gives the opportunity to read them again and often makes it possible to find where the problem is located...  
  
@++;  
Gauthier.