fra589/grbl-Mega-5X #152: Homing problems with a Core XY

PurpleCypher, 2020-11-13:
Hello, first of all, great job with this project, kudos.
I am building a remote controlled syringe consisting of  a CoreXY, a Z linear stage attached to it, and another really small A linear stage in the end of the Z. So basically I need to control a CoreXY and two other linear axis.

The movement controls fantastically, however there are two mayor problems on the homing front:
First, and this is the main issue, I am homing Z then A then X then Y. When CoreXY is not activated, both Z and A home perfectly, but when i activate the Core XY kinematics, the debounce in the A axis is more than stated in the config (usually 1mm, with CoreXY enabled 2mm) and this leads to sometimes errors in the syringe positioning, it baffles me, do you have any idea as to what to test ?.

The second is a known problem trying to home CoreXY stages. it is reported here https://github.com/fra589/grbl-Mega-5X/issues/124 ,  however the value '89' given as a solution in that post does not work for me with the CoreXY+Z+A. Perhaps i would like to understand what is needed of the mask to try an find an adequate one, or a recommended procedure to test this as trial an error?.

Thanks for all your help. Best regards

fra589, 2020-11-15:
Hi @PurpleCypher,  
  
CoreXY homing problem is clearly a bug, and I planned to build a corexy test machine so I hope I will can find it and correct it.  
Unfortunately, this will take a time (and optionaly some money) so I only can ask you to be patient...  
  
For your first debounce problem, can you post your configurations? axis and homing definitions in config.h, Grbl's output of $I and $$ commands, ...

By the way, since your 4th axis is a linear axis paralel to the Z axis, it should be named "W". cf. http://linuxcnc.org/docs/html/gcode/machining-center.html#_linear_axes
  
@++;  
Gauthier.

PurpleCypher, 2020-12-11:
Im sorry for my late answer. 
I was working on a project and given that coreXY seems to be still buggy i decided to redesign my machine into another configuration. Thanks for your super fast reply and please let me know if there is something to help with when you work on the core XY issues.
Best Regards

closed 2022-10-20 by fra589