fra589/grbl-Mega-5X #153: No pulses on the axis pins

sharlq, 2020-11-16:
i have installed grbl-mega-5x on the arduino and I am using TB6600 drivers i have unlocked the movement disabled the soft limits and then I have tried to give commands to move the axis motors but they dont respond I can see the command sent on the G code sender and tried many G-code senders also tried to send the command directly through the Arduino IDE but the motors is not moving
Given that I have tried gribl-mega on the same arduino / drivers / motors and it was working

fra589, 2020-11-17:
Hi @sharlq,  
  
The [default pinout of grbl_Mega-5X](https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout) is not the same than the original grbl_Mega's default to match the RAMPS shield places.  
You probably haven't connected your drivers to the right place.  
  
Here is the default Arduino's pinout for grbl-Mega-5X ([defined in cpu_map.h](https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h)):  
| Axis number | Program index | Default axis name | Step pin | Direction pin | Disable pin |
| ---: | :---: | :---: | :---: | :---: | :---: |
| 1 | axis[0] | X | A0 | A1 | D38 |
| 2 | axis[1] | Y | A6 | A7 | A2 |
| 3 | axis[2] | Z | D46 | D48 | A8 |
| 4 | axis[3] | A | D26 | D28 | D24 |
| 5 | axis[4] | B | D36 | D34 | D30 |
| 6 | axis[5] | C | D49 | D51 | D53 |
  
@++;  
Gauthier.

sharlq, 2020-11-17:
thanx very much

closed 2020-11-17 by sharlq