fra589/grbl-Mega-5X #155: Endstops do not trigger hard limits

babyeatingpsychopath, 2020-11-24:
I have a 3-axis laser machine that was using a cheap GRBL board with DRV8825 carriers. I smoked an axis on the board due to the current limiting not being set from the factory.

I downloaded this codebase and installed it to my RAMPS 1.4 board (a4988 carriers) and Mega1280 and have some problems.

My setup:
Two limit switches per axis, normally closed, except one limit switch on the z-axis (max travel).  I also have a fixed probe to set my focal distance.
I have "invert limit" set, $5=1.  I put a jumper on the unused z-axis-minimum pins, simulating a working normally-closed limit switch.  Watching the pin states, each switch individually triggers its axis.  Homing works fine on X and Y.  Does not appear to work on Z all the time.  Sometimes the machine acts like "z" is always triggered.

Additionally, triggering the endstop (any of them) during any travel does not trigger a hard limits shutdown, even though hard limits are enabled.  $21=1.  $20=1 as well, but it seems like the machine will happily drive well out of its envelope.

fra589, 2020-11-25:
Hi @babyeatingpsychopath,  
  
By default in grbl, Only software limits are activated and hard limits are not.  
For the machine don't drive well out of its envelope, you need to configure the max travels of your axis ($130 to $135).  
Anyway, if you realy need it, the hardware limits can be activated in cpu_map.h by suppressing the comment charracter @line 321:  
`  //#define ENABLE_RAMPS_HW_LIMITS`
  
For your Z axis problem, since the logic of the program is the same for all axes, I rather think that your problem is on the hardware side.  

@++;  
Gauthier.

babyeatingpsychopath, 2020-11-25:
Z-axis was caused by interference. Thanks for the tip.

Enabled the hardware limits, and all motion except for homing caused hard limit fault.  Tried both states of $5, with no change in behavior.

Ended up commenting out limits.c lines 203  and 212. Behavior works as expected now.  Diff attached.

[limits.diff.txt](https://github.com/fra589/grbl-Mega-5X/files/5596975/limits.diff.txt)

fra589, 2020-11-27:
Hi @babyeatingpsychopath,  
  
Your homing behavior look strange as it seems to work for me.  
But you are right, the lines 203 and 212 were leftovers from an old limit detection method that required processing the result of the limit only on the right bit. Hence the use of double negation (`pin = !! pin;`).  
This is no longer needed today, this will be fixed in the next version.  
Thanks for pointing this.  
  
@++;  
Gauthier.

closed 2020-11-27 by fra589