fra589/grbl-Mega-5X #156: How to Ramps E0 for 4th axis

boxgame9, 2020-11-26:
I have set the arduino with the correct firmware and tested the board.  I was surprised, after reading the write up that E0 is default for the 4th axis, to find that the y axis is cloned with E0.
As a novice I am at a loss to find a way forward. 
Is there a link someone can point me to for a resolution to this please.

fra589, 2020-11-27:
Hi @boxgame9,  
  
> I have set the arduino with the correct firmware and tested the board. I was surprised, after reading the write up that E0 is default for the 4th axis, to find that the y axis is cloned with E0.
  
???  
What is the "correct firmware" for you? Where and how did you get it?  
There is no cloned axis by default in my version...  
The standard axis definition in the [original fra589/grbl-Mega-5X firmware](https://github.com/fra589/grbl-Mega-5X) (it's defined in the config.h file) is: 5 axis, 3 linear X, Y and Z connected at their RAMPS place and 3 rotational A and B connected respectively to E0 and E1 RAMPS slots.  
  
> As a novice I am at a loss to find a way forward.  
> Is there a link someone can point me to for a resolution to this please.
  
Good points for beginning are the Grbl documentation WiKi:  
- The original one : https://github.com/gnea/grbl/wiki
- And the Mega-5X one : https://github.com/fra589/grbl-Mega-5X/wiki

@++;  
Gauthier.

boxgame9, 2020-11-27:
Hi Gautier
Thanks for the quick reply.
A slice of the config.h as in the firmware you mention above, downloaded today.
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'Y' // Letter of axis number 4 (dual Y axis)
#endif

 I believe I would only have to change the name 'Y' to 'A' for it to unclone. is this correct please.

Thanks for the links to the WIKI's
Lesli

boxgame9, 2020-11-27:
Hi Gautier.
I have just re downloaded it and the config.h now reads as I expected.  Not sure what went on there.
I will reload the firmware and leave a comment for you.

Thanks

Lesli

fra589, 2020-11-27:
Hi Lesli,  
  
I repeat: I don't know where you have downloaded your wrong config.h.  
The default of AXIS_4_NAME have always been set to A.  
[image]

Perhaps you should make a new download of all the latest original grbl-Mega-5X firmware version to avoid another error in another file...  
https://github.com/fra589/grbl-Mega-5X/archive/edge.zip
  
@++;  
Gauthier.

boxgame9, 2020-11-27:
Hi Gauthier
I have got your firmware and checked it for the4thaxis labelled A and yes this is so.
I have  uploaded the firmware and the A Axis does not run.
I have attached some pics that seem to indicate that the Arduino IDE is trying to compile for an Uno, strange.
I have deleted all the files in the documents/arduino/library folder so there is nothing there for the IDE to look at.

Thanks for your help

Lesli

[image]
[image]
[image]

fra589, 2020-11-28:
Hi @fra589,  
  
The mechanism to build the firmware from the Arduino IDE use a very simple sketch (grbl/examples/grblUpload/grblUpload.ino) which have not be modified from the very first version of Grbl on Aduino Uno. This is why the comment in the beginning of the file tell about "328p-based Arduino". But, it's only a comment. 😃   
This sketch is a C program that have only one useful line:  
`#include <grbl.h>`  
wich call the whole Grbl source as a library.
Carefully read the "Compiling Grbl" procedure at https://github.com/gnea/grbl/wiki/Compiling-Grbl.
  
Yes, the `$I` and `$$` output always help to understand more rapidly.
Especially, with the last up to date version version and default configuration,  your $I output should be:  
```
$I
[VER:1.1m.20201118:]
[AXS:5:XYZAB]
[OPT:VNMH,35,255,112]
ok
```
and not that I see on your `$I` output.
This tell me that it's probably due to the fact that an other version of Grbl stay in your Arduino library folder.  
So, before carefully following the [procedure for compiling and uploading the correct firmware](https://github.com/gnea/grbl/wiki/Compiling-Grbl), make sure to remove all the other Grbl's folders from your Arduino library folder (see https://www.arduino.cc/en/hacking/libraries for details about Arduino libraries).  
  
@++;  
Gauthier.

boxgame9, 2020-11-28:
Morning Gauthier.
Bonjour Gauthier.

I have found the compiling issue and I find it strange.
J'ai trouvé le problème de compilation et je trouve cela étrange.

I have two computers with windows 7 installed, one for design and a 3D printer the other for operating the cnc.
J'ai deux ordinateurs avec Windows 7 installé, un pour la conception et une imprimante 3D, l'autre pour faire fonctionner le cnc.

I have been compiling the mega 5x file on the cnc machine and as you know I have had problems.
J'ai compilé le fichier méga 5x sur la machine cnc et comme vous le savez, j'ai eu des problèmes.

I compiled the file on the 3D printer machine and it compiled with no warnings or errors.
J'ai compilé le fichier sur l'imprimante 3D et il a été compilé sans avertissement ni erreur.

Now I just have a small problem which I will investigate over the weekend.
Maintenant, j'ai juste un petit problème sur lequel j'étudierai ce week-end.

I will advise on the progress

Have a good weekend

Je conseillerai sur les progrès

 Bon week-end

Lesli

boxgame9, 2020-11-29:
Good evening Gauthier
Bonsoir Gauthier

I had two extra issues, both concerning movement.
J'ai eu deux problèmes supplémentaires, tous deux concernant le mouvement.

The first was an error code 15 showing on screen with no movement.  A bit of research showed me that error 15 was a soft limit issue.  I sent a $20=0 (disable the soft limits) command and the error disappeared .
Le premier était un code d'erreur 15 affiché à l'écran sans mouvement.  Un peu de recherche m'a montré que l'erreur 15 était un problème de limite souple.  J'ai envoyé une commande $ 20 = 0 (désactiver les limites souples) et l'erreur a disparu.

The second issue was still no movement???   I checked the code and everything I could still no movement.  A brainwave then told me to check the properties for the arduino in the devices menu.  USB speed 9600, whoops. Changed that and all now works as you describe it.
Le deuxième problème était toujours pas de mouvement ???  J'ai vérifié le code et tout ce que je pouvais toujours pas de mouvement.  Une onde cérébrale m'a alors dit de vérifier les propriétés de l'arduino dans le menu des appareils.  Vitesse USB 9600, oups.  Cela a changé et tout fonctionne maintenant comme vous le décrivez.

Thanks for you help

Stay safe

fra589, 2020-11-29:
Bonsoir Lesli,  (English translation after my French motherlang 😃 )
  
Sauf si tu sais exactement ce que tu fait et que tu es sûr que tes programmes ne sortirons jamais de l'enveloppe mécanique de ta machine, ou alors que ta machine ne possède pas de capteurs de fin de course pour le homing, desactiver les limites logiciel (`$20 = 0`) n'est jamais une bonne solution.
La bonne méthode est de configurer correctement les trajets maximum de chaque axe pour que le code erreur 15 corresponde vraiment à une erreur.
Il faut savoir que Grbl fonctionne de manière standard comme pratiquement toute les fraiseuses à commande numériques industrielles :
- Les buttées de prise d'origine sont situées du coté maximum des axes, ce qui permet de dégager au maximum l'espace de travail lors de cette prise d'origine (ça peux quand même être changé dans la configuration en fonction de la machine réelle),
- Lors de la prise dorigine, les coordonnées des axes sont mise à zéro (0), ce qui fait que tous les mouvements en coordonnées machine (G53) se font dans l'espace négatif.
- Il faut définir l'origine pièce (G54 à G59) ou un décalage d'axes (G92) pour travailler avec des coordonnées pouvant être positives et/ou négatives suivant la position de l'origine recalée.

Pour ton problème de liaison série, effectivement, la vitesse par défaut de Grbl est réglés à 115200 bauds :-)
Ca peut se changer à la ligne 48 de config.h
`#define BAUD_RATE 115200`

----------
English version
----------

Good evening Lesli,
  
Unless you know exactly what you are doing and you are sure that your programs will never leave the mechanical envelope of your machine, or when your machine does not have limit switches for homing, deactivate the limits software ($ 20 = 0) is never a good solution.  
The correct method is to correctly configure the maximum travels for each axis so that error code 15 is truly an error.
You should know that Grbl works in a standard way like almost all industrial CNC milling machines:
- The homing switches are located on the maximum side of the axes, which makes it possible to free up the working space as much as possible during homing (this can still be changed in the configuration according to the real machine),
- During homing, the coordinates of the axes are set to zero (0), which means that all movements in machine coordinates (G53) take place in negative space.
- The workpiece origin (G54 to G59) or an axis offset (G92) must be defined to work with coordinates that can be positive and / or negative depending on the position of the workpiece origin and / or the axis offset.

For your serial link problem, indeed, the default speed of Grbl is set to 115200 baud :-)
This can be changed on line 48 of config.h
`#define BAUD_RATE 115200`

Bonne continuation avec Grbl-Mega-5X,
Good continuation with Grbl-Mega-5X,

@++;
Gauthier.
P.S. N'oublie pas de clôturer ce sujet si toutes tes questions ont obtenu une réponse.
P.S. Don't forget to close this topic if all your questions have been answered.

boxgame9, 2020-11-30:
Bonjour Gauthier

Thanks for the reply.
Merci pour la réponse.

My machine has no limit switches fitted as I am slowly making it up as I am building the machine. 
Ma machine n'a pas d'interrupteurs de fin de course car je la fabrique lentement pendant que je construis la machine.

I believe that when the machine is switched on the homing settings are random until the homing switches are triggered. 
Je crois que lorsque la machine est allumée, les paramètres de référencement sont aléatoires jusqu'à ce que les commutateurs de référencement soient déclenchés.

I have not changed the work are size yet so if the random figures were bigger than those set on the configuration file it may already have been out of range giving me the error 15.
Je n'ai pas encore changé la taille du travail, donc si les chiffres aléatoires étaient plus grands que ceux définis dans le fichier de configuration, il se peut qu'il soit déjà hors de portée, ce qui me donne l'erreur 15.

Line 48 in the config.h is as you left it, 115200.  The windows properties on the computer were set to 9600.  After I altered the windows setting it all worked. 
La ligne 48 dans le config.h est telle que vous l'avez laissée, 115200. Les propriétés de Windows sur l'ordinateur ont été définies sur 9600. Après avoir modifié les paramètres de Windows, tout a fonctionné.

I have to practice my French to use it when I visit my house in Haute Vienne or as it is now Nouvelle Aquitaine.
Je dois pratiquer mon français pour l'utiliser lorsque je visite ma maison en Haute Vienne ou comme c'est maintenant la Nouvelle Aquitaine.

Once again many thanks for your help and will now close the issue. 
Encore une fois, merci pour votre aide et nous allons maintenant clore le problème.

Cordialement 

Lesli

closed 2020-11-30 by boxgame9