fra589/grbl-Mega-5X #158: Probing routine

arlimbad222, 2020-11-28:
Hi just made 6mm probe for my machine & now I want to zero work offset at centre of the circle can anyone help me out for this I am very new to probing . 😔

fra589, 2020-11-28:
Hi,

You need to use GCode probe movement `G38.2` => Grbl will stop the movement and immediately feedback of the probe coordinates through an automatically generated message when the probe signal is detected,
After the necessary number of probes to find your circle center, calculate it's coordinates (simple trigonometry),
Use `G10 L20 P...` to set the coordinate system depending on the result of your calculation
  
Normally, this is the Grbl interface program that should do this job.

@++;
Gauthier.

arlimbad222, 2020-11-28:
I generally use this method to zero work offset at corner of the rectangular part . But now it's all different thing I want to touch x+ x- , y+ y- & then 1/2 of the measurement and set the offset to zero . I'll drop a youtube link please open in you'll get it what I want I think it's possible but I am very new to probing that's why I need help .

https://youtu.be/_ZDlyLI_jbc

fra589, 2020-11-30:
Hi @ arlimbad222,  
  
That's what I said upper, so, here is a more detailed procedure... 
Only with Grbl and GCode:  
- jog your probe near ce center of the circle,  
- use the GCode G38.2 with sufficient X+ movement for that your probe touch the border of the circle,
- when the probe comes into contact with the edge of the circle, it stops automatically and Grbl returns the machine coordinates of the point where it stopped, store the  X+ value,
- use another GCode G38.2 with sufficient X- movement to touch the opposite X- border of the circle, then store the X- value,
- place the probe again near the center, then repeat for Y+ and Y-,
- calculate the X and Y coordinates of the center:  Xc = ((X+) + (X-)) / 2,  Yc = ((Y+) + (Y-)) / 2,
- jog to the exact center machine coordinates,
- place your new G54 (or G55...G59) X and Y piece's origin at this point with the GCode sentence: `G10 P1 L20 X0 Y0`
  P1 is for G54, P2 for G55 and so on, you can also use P0 for the current active G54...G59 reference.  
  
There is some software who automatically do this. For example, UGS have a plugin for it, see this simple demo: https://www.youtube.com/watch?v=kpz-6x_synE
I found this in less than 5 minutes with Google, there is probably many other.  
  
@++;  
Gauthier.

arlimbad222, 2020-12-04:
Hey far thank you so much for reply

i write some probing routine for corner zero i use 6mm probing tool for this but my question is not find corner of the part i want to touch my probe 4 side of hole and automatic find center of the circle .

Here is my code to find front right corner of the square part . i just need to set my tool approx corner of my part and boom this code will find very accurate corner . so my question is can i write this type of code for find center of the hole or any circular part .

G91 G38.2 Z-20 F30

G90 G10 L20 P0 Z0

G91 G0 Z2

G91 G0 X-5 ←

G91 G0 Y-10 ↑

G91 Z-5

G91 G38.2 Y15 F30 ↓

G90 G10 L20 P0 Y-3

G91 G0 Y-3↑

G91 G0 X15 →

G91 G0 Y15 ↓

G38.2 X-15 F30 ←

G90 G10 L20 P0 X3

G91 G0 X2 →

G28 G91 Z0
G90
G28 G91 X0 Y0
G90

fra589, 2020-12-05:
Hi @arlimbad222,  
  
You can't use this type of code to find center because GCode have no function to make calculation.
To find X center, you need to make 2 touch probes on X- and X+, store the touched values, then make calculation of  Xc = ((X+) + (X-)) / 2 and don the same for Y.  
  
As GCode, and so, Grbl have no calculation functionality, this have to be implemented in the GCode control panel / sender software.  
  
@++;  
Gauthier.

arlimbad222, 2020-12-05:
Oh thanks for reply you have any suggestions which g code sender will do this thing for me ?

fra589, 2020-12-05:
Hi @arlimbad222,  
  
Multiple searches of CNC probe software on Google give few software essentially no free...  
  
[Universal-G-Code-Sender (UGS)](https://github.com/winder/Universal-G-Code-Sender) control panel seems to have a touch probe plugin (see https://www.youtube.com/watch?v=kpz-6x_synE), since UGS is open source on Github, perhaps you can use it as a base to implement your need...  
  
I plan to implement this feature in the future of my own cn5X++ control panel software, but I don't know when I will find the time do do it.  
  
@++;  
Gauthier.

arlimbad222, 2020-12-05:
Oh man please find a time I am from India & I am planning to launch a new generation of cnc machine works with any android devices .

Like cnc with probing 4th axis some easy clamping system & yes it's aluminium frame cnc machine I am working on video when it's done I will share you .

Well UGS doesn't support center probing so I am waiting for your grbl & if you are working professionally I want to make my own gcode sender if you are interested mail me .

arlimbad222, 2020-12-05:
Hi@fra589 

https://drive.google.com/file/d/1ONxPMdMrM4AidGbS5Bq8mW6Er44aNse9/view?usp=sharing

Open this link here is small working video of my machine there is some more on process .

arlimbad222, 2020-12-07:
Hi @fra589 

Can you make one personal gcode sender for my company ?

fra589, 2020-12-09:
Hi @ arlimbad222,  
  
I thank you for the interest you have in my work, but it is not possible for me to accept your request.  
I have now retired for a little over a year, and the specific conditions of my retirement prohibit me from having a professional activity.  
  
After a higher initial training in mechanics and robotics, I spent most of my career in the IT department of a large company in the aeronautics industry. I loved the different IT jobs that I had the opportunity to practice, and today, I want to share with everyone the knowledge and experience that I have been able to accumulate during my career. That's why I got involved in writing open source and free software.  
I also have a lot of other great activities and unfortunately the days only have 24 hours. This implies that the evolution of my programs is a bit slow...  
  
The license of the cn5X ++ software you are interested in is currently GPL v3 (https://www.gnu.org/licenses/gpl-3.0.en.html), which implies that: You can run the software for any what use, you are free to study the operation of the program and to adapt it to your needs, you are free to redistribute copies. In the event that you redistribute the modified software, you must share the modified versions with the community under the same license.  
  
I therefore suggest that you use cn5X++ (https://github.com/fra589/cn5X) for your business.  
  
I could help you, as much as possible and in my spare time, for the addition of the functionalities and the evolution that you would like to see added.  
  
I cannot receive a salary for this work, but I can still accept donations, which you think is fair for the work that I would provide you. This allows you to buy the material to continue this activity (see the bottom of the main page of cn5X++ project (https://github.com/fra589/cn5X)).
  
@++;  
Gauthier.

fra589, 2020-12-12:
Hi @arlimbad222,  
  
GrblGru have probing functionality, I think it's what you need.
See: https://www.youtube.com/watch?v=DPtNWnFX6jE
  
Free download if it here: http://GrblGru.com  
  
@++;
Gauthier.

arlimbad222, 2020-12-12:
Oh thankyou so much I'll see the video soon i am out of town right now will update you soon

closed 2021-01-11 by arlimbad222