fra589/grbl-Mega-5X #159: SD-card + lcd2004 support from bdurbrow/grbl-Mega ??

erosenst, 2020-11-29:
Is it posible to merge some bdurbrow code to get standalone grbl-mega-5X ramps 1.4 controller?

I have plasma cutter project XXYZA axis and I'm intrested also THC codepiece throug bed temp input. Pipe cutter axis (A) is not done yet.

erosenst, 2020-11-29:
https://github.com/bdurbrow/grbl-Mega

fra589, 2020-11-30:
Hi @erosenst,  
  
It's perhaps, and probably possible, but this doesn't match the grbl-Mega-5X project philosophy.  
In this project, I try to maintain a maximum optimization of the code as close as possible to the basic hardware and its native language. This why I use standard C and not C++, this is also why I prefer not to add functions which are not directly used for the functions of the CNC machine.  
Unfortunately, I don't have the time to do it today, but I already thought about making a device allowing the autonomous use of my CNC milling machine. But this device will have to use another Arduino board or another hardware cheap card to manage the LCD screen, the SD card reader and a set of buttons to perform all the jog movements, the homing functions and the definitions of machine origins and offsets. Thus, the Grbl card could remain as efficient as possible without having to do any work other than controlling the motors and sensors.  
This will therefore be another project that could come in addition to Grbl as a non-compulsory option.  
  
If you really want this system on grbl-Mega-5X, you can always develop it yourself. I could try to help you if you have simple and precise questions for which my knowledge of Grbl internals could have quick answers.  
  
@++;  
Gauthier.

erosenst, 2020-11-30:
Thanks for fast reply. You think reading g-code from sd-card is performance issue vs. serial port handling? Can it done in idle cycles? How 3D-printers do it? LCD can update when idling, when machine works display update is not needed?
I understand C vs. C++ broblem so it needs massive translation to use bdurbrow code..

How about plasma THC is it posible to add in near future? It is machine operation code and I think there are some example codes in net..?

fra589, 2020-11-30:
Hi @erosenst,  
  
If we decide to read GCode from an sd-card, this doesn't exempt us to continue to read, write and deal with the serial port, so, there will be more job to do.  
When we read GCode from any input (serial or sd-card), there will be no idle time as long as the complete read GCode program will not been completely executed.  
  
The internal operating principle of Grbl is as follows:  
- The serial port is read by a hardware interrupt which simply places the received character in a pending buffer. This can therefore happen at any time. This is the processing code for this interrupt which calls the routines corresponding to the real-time commands,  
- The main program loop reads the serial buffer character by character and starts the interpretation of this line upon reading of the end of line character (CR or LF) and returns the result to the serial port (OK , Error or Alarm),  
  - If the line starts with $, we interpret as system command,  
  - otherwise, we analyze and execute a GCode command.  
- The execution of a GCode movement is not done in real time:  
  - The movement is calculated for it's displacement, speed and acceleration,  
  - in order to be able to control it with it's speed and acceleration, the displacement is broken down into as many motor steps as necessary and the moment at which the impulses are to be sent to the motors are calculated.  
  - The result of the calculation is put on hold at a place called the "planner" and it is from this place that the impulses are sent to the motors using a timer interrupt. 
   
The processing time of the processor must therefore be divided between the 3 main processes which are:  
- The processing of the serial port,  
- Treatment of motor impulses,  
- The decoding of the GCode.  
  
The first 2 processes must remain with high priority and light enough to guarantee that the treatment of the material interrupt will be completed before a new call.  
  
I don't have any experience about plasma THC, but it should not be very difficult to adapt the spindle management code to drive it since this code aleready can manage PWM output and enable + direction logical output pins. Replacing Spindle by plasma should be simple.  
  
For output, updating the active coordinate output while moving is also a very most important thing, this is why the "?" Grbl command is a real time command wich not wait other to be executed. Most of Grbl interface use this to output coordinates almost continuously. This allows to monitor the position while jogging or during working travel. The use of an LCD display would double the work for output.  
  
The use of 2 boards will not be "2 boards for one only standalone solution", but 2 boards for 2 complementary solutions:
Board 1 for the CNC machine management, board 2 for contrôling the board 1 and display it's status... There are many cards in a single computer.  
  
Today, 3D printers increasingly use more powerful cards than the Mega 2560 because they want to do too much at once and small 8-bit cards like the Mega are no longer enough. As a result, it becomes more expensive to buy and complex to maintain.  
  
I prefer to stay on [the KISS principle](https://en.wikipedia.org/wiki/KISS_principle) 😃  
  
@++;  
Gauthier.

erosenst, 2020-12-03:
Hello about THC. Have you any comments if trying to loan THC functions from https://github.com/UnfinishedBusiness/XmotionFirmware/tree/master/src
main.c
and try to port them 2560 A3 & A4 pin ..?

Today I make first cuts with my cutter table with Mega-5X. Works fine but lacks THC. HF starts works if nozzle close enough metalsheet but too far it crash grbl..

Btw. have any ideas why dxf2gcode produced G2 & G3 dosn't run but give error:33 ? Any examples G2/3 x y i j that works? Currently grbl run G2/3 x y r only??

fra589, 2020-12-05:
Hello @erosenst,  
  
I'm sorry, but I really don't know anything about TCH plasma cutting and have no needs for it. So, I couldn't be of much help in helping you implement this feature.  
I took a quick look at the link you put, given my ignorance about plasma cutting, I'm not able to comment on how the program works.  
On the other hand, in terms of form, It would not seem appropriate to me to have added the thc functions in the main.c file. This is not a good practice, it does not help for maintenance or to allow other programmers to study open source code and participate in its evolution.  
If you want to take inspiration from this program to complete grbl-Mega-5X, I advise you to take care about the organization of the code and add one (or more) specific module(s) using the additionals files new_module_or_functionality.h & new_module_or_functionality .c. You should also add the necessary options for its activation / deactivation and its parameters in a new section of the config.h file.  
  
@++;  
Gauthier.

erosenst, 2020-12-07:
I tested to add some thc code to grb-mega-5x. It compile fine and give some voltmeter to status report. It works with analog input from ramps1.4 temp1 input so it need to unsolder 4.7kohm R25 resistor or you got max readings.. It connects plasma with 10Mohm + 150kohm so 0.015x divisor to actual voltages. 1.1v ADC reference.

Question if I have thc_z_up and thc_z_down booleans what is correct place to make steps by them? (ISR or some else place?) 

I have idea that spindle speed is plasma target voltage and it moves Z only when M8 is active and plasma voltage over 40v (=S560)

----clip----
report.c:

  // Report current machine state and sub-states
  serial_write('<');
  #ifdef PLASMA_THC
  print_uint8_base10(analogVal >> 4); // close enough
  serial_write('v');
  serial_write('|');
  #endif

main.c:

#ifdef PLASMA_THC
//volatile uint16_t analogVal;
//volatile uint16_t analogSetVal=838; // 0.9v / 0.015 = 60v, 10mohm / 150kohm
volatile uint16_t analogSetVal=0; // thc off
bool thc_z_up=false;
bool thc_z_down=false;

ISR(ADC_vect){
  // Must read low first
  analogVal = ADCL | (ADCH << 8);
  if (analogSetVal > 700) { //THC is turned on, 50v border
    thc_z_up = false;
    thc_z_down = false;
    if (analogVal > (analogSetVal + 10)) { // +0.7v  hysteris
      thc_z_down = true;
    }
    if (analogVal < (analogSetVal - 10)) { // -0.7v  hysteris
      thc_z_up = true;
    }
  }
}
#endif // thc

  sei(); // Enable interrupts

#ifdef PLASMA_THC
  /* ADC Setup */
  ADCSRA = 0b10111111; // adc on 7, start 6, auto 5, int 4, int 3, xtal division 128 2:0

//  ADMUX = 0b10000110; // 1.1v ref 7:6, rightH 5, mux 4:0 -> A14 pin (bed temp)
  ADMUX = 0b10000111; // 1.1v ref 7:6, rightH 5, mux 4:0 -> A15 pin (temp1)
  ADCSRB = 0b00001000; // - 7, ACME 6, - 5, - 4, mux5 3, free run 2:0

//  DIDR2 = 0b01000000; // A14 pin digital off (bed temp)
  DIDR2 = 0b10000000; // A15 pin digital off

  // and then we run
  ADCSRA = 0b11111111; // adc on 7, start 6, auto 5, int 4, int 3, xtal division 128 2:0
#endif // thc

erosenst, 2020-12-11:
There is current plasma THC code. It maybe work..(?) Atleast it compiles ok..
It need in main.c only ADCSRA = 0b11111111; after sei() to start ADC once.
And if like voltages in report.c lines from previous message..

S870 seems work fine to me then M3 and when end cutting S0 and M5..
Not sure dos it actually move head need more testing..

[stepper.txt](https://github.com/fra589/grbl-Mega-5X/files/5680353/stepper.txt)

erosenst, 2020-12-13:
Hello I added LCD+SDcard code to 5X. It compiles 77kb instead 36kb. Works at least some. One nc-file stopped to G38.2 when loaded from SD-card. Not know yet why. From usb same file executes normally.  Maybe you like to test it..?
[image]

[grbl-Mega-5XLT.tbz.zip](https://github.com/fra589/grbl-Mega-5X/files/5684652/grbl-Mega-5XLT.tbz.zip)

erosenst, 2020-12-13:
Find G38.2 broblem, it disable pullup from D49 (my probe pin) There is in cpu_map.h conflict in probe pin and control port.. Need to modify.

fra589, 2020-12-14:
Hi @erosenst,  
  
> It compiles 77kb instead 36kb
36 kb for all the Grbl milling functionalities, and 41 kb for only sc-card and lcd... This is a very good reason not to add this functionality to the main Grbl program.  
  
@++;  
Gauthier.

closed 2020-12-14 by fra589

erosenst, 2020-12-14:
Well fit g-code sender to 41kb including os..
It can be do more diet as bdurbrow code is but it was "ready" pack.. I think it dosn't affect performance much and there is in 2560 250kb room. 77kb is not much.
When you engineer second mega to send g-code, maybe you can do code way that it can be dropped to grbl too..? This bdurbrow code is quite easy to port standalone if you need codepiece to start.. It is c/c++ hybrid, sd-card piece is c and menu c++.

erosenst, 2020-12-19:
Updated code little. More compile options mainly to disable useless (for plasma cutter) functions/menus. Now compile 55.5kb.

[grbl-Mega-5XLT_v0.1.tbz.zip](https://github.com/fra589/grbl-Mega-5X/files/5720203/grbl-Mega-5XLT_v0.1.tbz.zip)