fra589/grbl-Mega-5X #161: Question about 5-axis machine disable limit switch properly and number of linear axis?

erosenst, 2020-12-02:
Hello I have 5-axis plasma cutter machine with cloned X axle and 5th axis rotational pipe cutter (5th not ready yet).
Machine is ramps 1.4 XYZX'A configuration so 4th axis is cloned X to E0 and rotational axis is E1.

What is correct number for linear axis value 3 or 4 in config.h?

I wound also compile problem when try disable E1 min/max limitswitch and max limitswitchs. Macro show undefined axis 4 & axis5 max limit. What is proper lines to disable max/min limitswitches?

I have configuration
        min             max
axis port bit pin port bit pin
0      E     5  D3    E   4  D2
1      J      1  D14  J   0  D15
2      D     3  D18   -
3      D     2  D19   -
4      -                    -
and Z max to probe pin aux3 D49 L0

Thank you

erosenst, 2020-12-02:
It is posible to compile defining all unused positions to aux3 D53 (B0) but it use one pin more. It is solution but not correct answer.

fra589, 2020-12-03:
Hi @erosenst,  
  
N_AXIS_LINEAR is only used for mm to inch conversion. If you only use mm as units, it's not so important.  
N_AXIS_LINEAR is referring to the axis number, not their names, so you need `#define N_AXIS_LINEAR 4`  
  
It's useless to disable the E1 min/max limits switchs, you can simply don't connect them, and so, don't use them. If you really want to free the E1 axis limit affected pins, you can define them to the same pins than those used by another axis.  
  
@++;  
Gauthier.

closed 2020-12-08 by erosenst